vlachoudis/bCNC #407: Engraving with Autolevel -

Guenni75, 2016-08-18:
Hi, I have a Woodplate (450x450mm).
I ran an Autolevel probing cycle with 10x10.
All works nice.
I set the Zeropoint at the middle of the plate and set the V-Mill to Z=0 at this Position on Top of the Material.
I engraved about 10 single Letters in a semicircle.
The farther the job had progressed, the thinner the Letters are.
I stoped the Job and checked the Z-Axis. 
At Z=0 the V-Mill had 0,5mm to the Material!
I set Z again on the Material and ran the last letter again. It really goes deeper then before.

After that i engraved the Letters one by one (the Editor is great!) with checking Z between each of them, the height was correct every time.

It seems that after a while the calculation of the  Z coordinates with the Autolevel values isn't correct (rounding?).

Effer, 2016-08-18:
Hi @Guenni75 ,
if you air cut the same job, does the Z axis shift the same?
Can you attach a Dial Micrometer and see if there are any mechanical loose?

With some up-cutting mills I've encounter the same behavior.

Guenni75, 2016-08-23:
The same job without autolevel works perfect.
First I thought the mill slips into the collet.
I marked it with a pen and there was no slipping.
I have a direct belt drive. The coupling of the stepper is tight.

Guenni75, 2016-08-23:
I think it is a problem with the Mikrostepping.
If there are to much values that have been rounded in the same way (because the stepping is not small enough), this could happen.

closed 2017-02-14 by vlachoudis