vlachoudis/bCNC #408: lower Case G-Code from cnc.py

TSpielmann, 2016-08-19:
Tools using cnc.py (like Text or Bowl) create G-Code with LowerCase Commands. 
I m using bCNC with Smoothieware  edge-1aa47c6, Build date: Aug 17 2016 04:05:43 and got Command Errors. 
Patching cnc.py to produce UpperCase Code and modifing the Plugin Code to Output UpperCase initial Code (like "F" instead "f") solves it for me.
Don't Know if this is an bCNC or Smoothieware Issue! (Sorry If wrong Place)
If this is an Issue, it hits the Plugins too.

vlachoudis, 2016-08-20:
It's true that bcnc generated most of the code in lower case while smothie needs upper case. However if you select the smothie board as a controller when the gcode will be sent to smoothie it will be converted to uppercase.

TSpielmann, 2016-08-21:
That's right and was my Mistake. (not switched to Smoothie)
I beg your pardon!

closed 2016-08-21 by TSpielmann