vlachoudis/bCNC #411: 99% does not seem to finish

atrueresistance, 2016-08-26:
New install from github on a chromebook running chromium. Completed is never truly complete. It never goes past 99%

commit `88fc179effcf5206fa109a8dbde3b49491ced166`

[image]

[Tea Sign.nc.zip](https://github.com/vlachoudis/bCNC/files/440332/Tea.Sign.nc.zip)

Effer, 2016-08-29:
Short answer: it's the M30 command at the end of your part program, remove and it will end nicely.
 It's just a metter of your postprocessor (how much I hate this phrase)

M30 is and "End program" command (From shapeoko wiki)

> End and rewind. Rewinding refers to programs on punched paper tape. Shorter programs would be connected into an endless loop and be terminated by an M2 (see above), while longer programs would of necessity be rewound at their end by the M30 command.
 **On modern machines M2 and M30 may be considered equivalent.** 

The big question is: could we simply drop this command? Is grbl newest version capable of menage this?
@vlachoudis ? @chamnit?

(FYI  I'm still using version [0.9i.20150315:])

atrueresistance, 2016-08-29:
I am using 0.9g from Carbide3d. Have not tried J grbl. 

The weird thing is this doesn't always happen, and I'm not sure how to isolate. For instance the day that this was happing it did it fairly consistently. Yesterday I queued up and ran 10 different engravings using autolevel and it always completed. So not sure... 

IMO the code runs and completes, just the timer keeps running so it isn't an issue.

vlachoudis, 2016-09-04:
Do you use the tool change functionality?

ErikHGV, 2016-09-11:
I have seen some G code ending with M30 but no CR+LF (new line) causing the M30 to be read as M3 (spindle on) in the bCNC.

chamnit, 2016-09-11:
This is caused by a subtle change in how M30/M2 program end works after Grbl v0.9g, where it forces a feed hold. Newer Grbl versions no longer do that. You'll need to send a resume real-time control character to complete the job to 100%, but it's already done.

closed 2016-09-15 by vlachoudis