vlachoudis/bCNC #412: [Flatten Plugin] - Generated GCode file don't respect the settings

onekk, 2016-08-28:
When I use flatten plugin and want to save the file, the block that define the "raster border" is generated and saved in the GCode produces.

I noted that it is defind with (Block-enable: 0)

I can guess that bCNC will not execute this code, but if you rely on the generated code and use it on another "sender" it will not do what you expect.

The questions are:

1) I guess correct?
2) if 1 is yes there is a manner to avoid this without manual delete the block in every generated file?

vlachoudis, 2016-08-28:
1. yes
2. No.

Unfortunately is a way in bcnc to add metadata in the gcode

onekk, 2016-08-28:
OK Many thanks.

closed 2016-08-29 by vlachoudis

Effer, 2016-08-29:
There was another user, some time ago, who asked for a way to export the modified gcode without all the metadata (maybe to the clipboard?). Maybe we could even add an option to omit the disabled gcode from that export.

onekk, 2016-09-01:
This solution will be good, a save option in the editor tab or maybe a "save Gcode" in file tab that trigger a save window when you could choose some the option like "don't save disabled block" and maybe "save gcode in uppercase", I know grbl will accept even lowercase, but almost all GCODE i've seen is uppercase.

Regards

Effer, 2016-09-06:
@onekk latest commit #416 has the capability to save gcode as plain text. (only enabled block in uppercase)

onekk, 2016-09-09:
Thanks, i read only now, i will try maybe monday :-D