vlachoudis/bCNC #414: After closing and reopening bCNC the status is not shown

onekk, 2016-09-01:
Program     : bCNC
Version     : 0.9.6
Last Change : 24 Aug 2016
Platform    : linux2
Python      : 2.7.9 (default, Mar  1 2015, 13:48:22) 
[GCC 4.9.2]
TkVersion   : 8.6
TclVersion  : 8.6

These above are the relevant info from the bug report form in the program.

The program is close with the quit icon

when reopened it show "Not connected" and i can't send anything to the nmachine, closing and reopening the port won't help.

But if i exit and from the console use minicom to communicate with the machine grbl is active and i can't send the commands.

System is debian jessie (Armbian) running on an Olimex Olinuxino A20 (armhf)

Regards

Carlo D.

BlueOrangeLive, 2016-09-01:
@onekk 
After close and reopen everything is OK here.
Downloaded and installed today on my Testsystem.
GRBL = Version 1.0c. vom 30.03.2016 
Program     : bCNC
Version     : 0.9.6
Last Change : 24 Aug 2016
Platform    : linux2
Python      : 2.7.12 (default, Aug  9 2016, 15:48:18) 
[GCC 6.1.1 20160621 (Red Hat 6.1.1-3)]
TkVersion   : 8.6
TclVersion  : 8.6

My System is fedora 24

Regads
 Jürgen H.

vlachoudis, 2016-09-02:
@onekk maybe there is a process still haning and using the serial port. You can check the processes using a file with the ```lsof``` command like
```lsof | grep /dev/ttyUSB0```

onekk, 2016-09-02:
`lsof | grep /dev/ttyUSB0` give me no output, but i haven't say one thing,

when i reconnect, my cnc "bump" as it do when the serial connection is established, (and in effect it happens even when i try to connect with minicom).

But the status line show "not connected" , but if i click on the status line bCNC tells me "connection with grbl established", but the status lis stuck to "not connected" in red and the DRO is blank.

I0m using the python-serial version 2.6-1.1 for debian jessie 8  (armhf) 

Thanks and Regards

Carlo

onekk, 2016-09-02:
Some news, it seem that pyserial 2.6 is affected by some bugs, i installed a virtualenv with python-pyserial 3.1.1 and everything work fine.

But the stock debian jessie is stuck to 2.6 version.

vlachoudis, 2016-09-02:
Thanks @onekk for reporting. I am closing the issue

closed 2016-09-02 by vlachoudis