vlachoudis/bCNC #415: Stuck in Idle state.

Protoneer, 2016-09-02:
On a few occasions I have left the machine on and when I come back bCNC reports that the machine is idle but when I send a movement command it does nothing. 

I am running the latest GRBL with with bCNC 0.9.5 on a RPI3 via hardware serial.
[http://wiki.protoneer.co.nz/Raspberry_Pi_CNC](http://wiki.protoneer.co.nz/Raspberry_Pi_CNC)

To resolve this I usually close the connection and reconnect. I also recall seeing that the position polling was not happening anymore.

vlachoudis, 2016-09-02:
Was there any error message in the console?

Protoneer, 2016-09-02:
Thanks. I did not check but will do so in the future...

I am fine tuning my system and also find that when I set the step/mm the
machine will some times just start moving in an unintended direction when I
click on the movement arrows the first time after setting it. Luckily my
machines is big and I can stop it before it smashes into an edge.

onekk, 2016-09-10:
python-serial version 2.6-1.1 for debian jessie 8 (armhf) is broken and give some problem in reconnection.

I installed python-serial 3.1-2~bpo8+1 from jessie backports and it works without problems

Maybe it will help

ruenahcmohr, 2016-11-19:
I have had this problem too, so far its on curves, there are no errors and the machine just stops part way thru a file, status idle.

vlachoudis, 2016-11-19:
Is it reproducible or random?

ruenahcmohr, 2016-11-20:
I'v not tested enough to know, it almost looks like it happens on the start of one of the blocks it breaks the nc file 
into (which is a feature I really love btw)
I will feed in updates as I learn about it.

Rue.

-- 

  -- -- -- -- -- CUT HERE -- -- -- -- --

The quote for any above mentioned products and or services is $1000000.00, taxes excluded, negotiable.

***********, Owner
http://ruemohr.org/
*************
(604) ***********
************
************** BC Canada
************

vlachoudis, 2016-11-20:
Do you use autoleveling? Its the only place that it breaks the arcs to linear segments.

ruenahcmohr, 2016-11-20:
nope.
so far its happened 3 times, out of about 20 runs, both times I'v been able to disable the other nc blocks (less header 
and footer) and run just the missing parts.
I should mention its happening with a machine running win XP (and cnc is using grbl, latest)

ruenahcmohr, 2016-12-10:
the last few problems have been outright lockup of grbl, I just tried it on linux and had the same things happen (before was windows xp) from what I can tell the sender crashed. looking at the serial module of python I can see why gui+serial isn't an easy task.

vlachoudis, 2016-12-11:
Could it be noise on the USB cable?

ruenahcmohr, 2016-12-12:
ok I'v put some more time in.
a) The problem exists on linux as well as windows
b) its not repeatable 
c) its something with the bcnc software.

I tried a linux machine and had a short job (just a half hour) stick in idle, going into a curve.
I restarted the job (zeros were still ok) and it ran just fine.
I ran a 15 hour program with the python script that grbl comes with in the docs and (aside from a fight with zeroing) had no issues.

vlachoudis, 2016-12-12:
Could you send the gcode, grbl settings (and version) and bCNC settings, to check if I can repeat on my test machine?

ruenahcmohr, 2017-01-26:
I keep meaing to do this, but keep forgetting to get the bcnc settings, I can confirm its not a machine problem as the spooler I wrote seem to always behave fine.  I will post the files yet.

vlachoudis, 2017-01-29:
I am closing the issue for the moment until we get extra information

closed 2017-01-29 by vlachoudis

rusk911, 2017-10-18:
I have such problem also. bCNC is loosing serial port in some way when the job is in progress, reconnectiing to ttyusb0 failing, I have to connect to ttyusb1, set up coordinates and continue job. Sometimes latest coordinates  received from grbl are outdated, that usually means broken part. UGS works well at same setup.
I use Linux Mint 17

ivandewolf1, 2020-07-13:
I have had this issue a couple of times, this time it was 5 hours into a 7 hour cycle, no idea what caused it. Next time I'll look at the console

ruenahcmohr, 2020-07-13:
I ended up writing my own spooler program
would you like to give it a try?

ivandewolf1, 2020-07-13:
I'd love to! what language are you writing it in? by any chance, might it
run on a raspberry pi under raspbian?
-Ivan DeWolf
cell: (778) 926-8392
-------------------------------------------------------------------------------------
"Perfection is achieved when nothing else can be removed"
-Yvon Chouinard

ruenahcmohr, 2020-07-14:
Its written in python, Its known to not be 100% problem free, but its working great for everything I do so far.

https://github.com/ruenahcmohr/TapeRunner

feedback is cool..... :)

Harvie, 2020-07-14:
> Its written in python, Its known to not be 100% problem free, but its working great for everything I do so far.

Sorry to hear that bCNC made you to take a course of such a desperate action :-) Given that now you have experience with handling GRBL communication in python, would you mind sifting through our codebase and suggesting what can be improved in order to achieve same level of satisfaction as you currently have with your spooler? :-)

https://github.com/vlachoudis/bCNC/tree/master/bCNC/controllers

ruenahcmohr, 2020-07-17:
My handshaking is crude, I wait for an OK before I send another line.
But I need to change the method I'm tracking it via into an object that can cover a few more bases.

I think I was having challenges following the bcnc source. I can give it a go in a while and see if I can cough up 
suggestions.

Buggy-Ben, 2021-04-24:
Has there been a solution to this problem? 
I have exactly the same problem mine is very frequent and does not happen with Carbide motion or UGS.