vlachoudis/bCNC #417: Editor

rennen11, 2016-09-05:
With some G-code I can`t see any geometrie in the geometrie window.
loading another file I can see geometrie.
Comparing the two codes shows no difference in commands, only in coordinates.
Running python 2.7 on Win7

Effer, 2016-09-06:
Can you post the code? You can zip and upload it here or use https://pastebin.com/

rennen11, 2016-09-08:
Hei Filippo,

 

Here is the code.

With other codes the behavior is also sometimes funny.

When I load a code after another, Istill see the picture of the previous part.

Clicking on the toggle display switch brings up the new one.

 

Another Question:

In the Open File Windows I have only access to C:\ , I am not able to change to disc B:

 

Many thanks to your help.

 

Rennen11

Effer, 2016-09-08:
Hi  @rennen11 ,

I can't see any file attached.
The behavior you describe is quite strange, try to change the "time out display" in the combobox on the top-right corner of the rendering canvas. Default value is 2 (seconds) after that time the drawing is suspended and a warning should appear on the bottom of bCNC window.

To navigate to another folder write the desired path in the open file dialog:

[image]

It seems there's no easy and cross platform way, to list all disks in python.

rennen11, 2016-09-08:
An other try.

 

 <https://cloud.githubusercontent.com/assets/1007007/18348884/4bdd4d6c-75ce-11e6-86ce-3b51b4f70764.png>

Effer, 2016-09-08:
No luck neither this time. I suppose you have to use the web page to upload the file, attaching with your email client doesn't work with github.

Does this happen after you select the geometrie and hit canc to delete it?

rennen11, 2016-09-08:
hope you can see the file now.
[SD7080_190_Bearb_1.zip](https://github.com/vlachoudis/bCNC/files/462114/SD7080_190_Bearb_1.zip)

rennen11, 2016-09-08:
hope you can see the file now.
[SD7080_190_Bearb_1.zip](https://github.com/vlachoudis/bCNC/files/462128/SD7080_190_Bearb_1.zip)

Effer, 2016-09-09:
Hi @rennen11 ,
got the file ;)

I think is just a matter of file format. The file you provide use the number of line at the begin (N10 .. N20 and so on). It's seems like bCNC doesn't like it. There's a function defined in the code (*removeNlines*) but it's never call and miss full implementation.
To add in in the TODO @vlachoudis ?

Effer, 2016-09-09:
@rennen11 I was wrong. bCNC support NCodes.
In your file there is the G94 code in line N140 that broke the parser. If move the GCode to line before everything look good

[image]

Effer, 2016-09-09:
It seems like method motionStart in CNC.py doesn't accept some expression that grb can work with.
As example any movement G0/1/2 that contains a G93 or G94 in the same line. It's accepted by grbl but broke the canvas rendering.

It's seem easy to fix

```python
elif gcode==91:
	if decimal == 0:
		self.absolute = False
	elif decimal == 1:
		self.arcabsolute = False

#new code for feedmode
elif gcode==93:
	pass

elif gcode==94:
	pass

elif gcode==95:
	pass
#end of new code for feedmode

elif gcode==98:
	self.retractz = True

```

But before a PR I prefer to wait @vlachoudis for clarification

rennen11, 2016-09-11:
If I put the G93 or G94 in an extra line, the rest of the code is also not shown.

But it`s not a problem for me because its not necessary for me to use it.

Thank’s for your efforts

vlachoudis, 2016-09-15:
@effer is correct. I've pushed an update

closed 2016-09-15 by vlachoudis