vlachoudis/bCNC #422: Delete elements broke sync between canvas and editor

Effer, 2016-09-08:
How to reproduce:
Start bCNC
Go to tools tab
* Close bCNC
* Start again bCNC (this time it should open directly to tool tab)
* Now create something with any tool/plugin (ex Flatten)
* Click and drag a selection area over the produce geometry (no single click)
* Hit canc to delete the geometry
* The geometry should't be erased
* Go to editor tab and see that the related gcode is really missing

See screenshot (not photoshoped!)
[image]

Maybe this is related to both  #417 #418 and #421

vlachoudis, 2017-01-29:
@Effer I finally found how to reproduce it and it is corrected now. The problem was that the global message that was send from the Delete button if the editor was not yet displayed (at least once) it was not intercepted. So now I am forcing all messages to be send to the master frame which is always there.

closed 2017-01-29 by vlachoudis

Effer, 2017-01-29:
Good job @vlachoudis, an hard bug to find? I've lost some times looking at code without result. Can you point me at the commit that fix it?

vlachoudis, 2017-01-29:
https://github.com/vlachoudis/bCNC/commit/3c1f0b1178a79484d1621c4ed5983fef64e8bd8a