vlachoudis/bCNC #423: I can't find ./bcnc file for custom Menu in raspberry pi

robcop2000, 2016-09-10:
Where is the ./bcn file in raspian weezy.
Thanks in advance

onekk, 2016-09-10:
The file is created in the home directory, the .bcnc.ini file is only a template, you have to create a .bCNC file in your home dir maybecopying bcnc.ini and rename it and do the customization you need,
it is generally created on the first start of bCNC, so you have to rename your customized copy and save it for further use.

Across versions it may became unusable.
 I use a bunch of customized buttons to preset some operation on my cnc and generally when changing bCNC versions i rename the old .bCNC in homedir with a name maybe .bCNC.old and then using a text editor i inspect the new .bCNC created by the new bCNC version and then stop bCNC and copy my customisations in place, notably the machine defs and the buttons settings, and maybe your preferred tool setting.

robcop2000, 2016-09-12:
I have seen in the Home directory and there isn't the .bcnc so I have modified the bcnc.ini and it works

Effer, 2016-09-12:
Hi @robcop2000 
try in this way:

[image]

Click on the config label to show this additional button

It should open your default editor or ask you how to open the settings file.
From there you can see where the file is saved.
Change to bcnc.ini will be overwritten if you update the software without keeping a backup.

robcop2000, 2016-09-12:
Found
Thanks I was searching with the file explorer.
Thanks for your help

closed 2016-09-12 by robcop2000

Effer, 2016-09-12:
Just for curiosity where it was?
Can you also add the info back in the Wiki? Maybe here https://github.com/vlachoudis/bCNC/wiki/OS#raspberry-pi

robcop2000, 2016-09-12:
It was in the Home directory, but is hidden.
At this link you can see two picture.
One looking in user as you said.
[](https://www.dropbox.com/s/b2cqdb64su3pej1/IMG_1933.jpeg?dl=0)
Another looking in file explorer
[](https://www.dropbox.com/s/tg6n56hqp1e6d5q/IMG_1935.jpeg?dl=0)