vlachoudis/bCNC #424: [Suggestion] make the .bCNC modular

onekk, 2016-09-10:
It was becoming diffficult to mantain the .bCNC in sync with customisation across version change.

In prevision of the 1.0 version, it will be advisable to make the .bCNC file contain only interface customisations, and put the toll, material and ather settngs in external file, maybe in separate direcory with meninful name (bCNC-settings?)

with maybe a toll.ini and material.ini and ,maybe a machine.ini

in this manner the relevant customisation will be preserved across bCNC version change and will be possible make a script to convert the tables between versions

I have used successfully in one of my project adopted this approach and it served me very well, if the program at the first startup din't find the appropriate directory it create them othewise it utilize the files present in the directory.

Regards

lalo-uy, 2016-09-10:
One option is to have the #include comand in the .ini file.
So any one can break it is as many includes as he wish.

onekk, 2016-09-11:
I'm trying to implement a modular configuration directory under ~/.config/bCNC/ directory in Linux or in $User/%AppData%/bCNC in windows.

But as the load config is fairly easy, you put the filenames in the line that read the config file and the last config variable readed is what it is used.

For saving it seems that all the configuration is saved .bCNC file.

I'm trying to guess from the code how to load the Buttons Machine and Tools configuration from different files and saving them in the same file when you perform a save or automatically at the program closing.

onekk, 2016-09-13:
After some look at the code of bCNC modifying the ini file .bCNC to split in different files will be a difficult exercise, without some discussion with @vlachoudis it is difficult to integrate in the structure of bCNC.

vlachoudis, 2016-09-15:
@onekk a modular approach with be certainly better. I didn't think of it at the begging that is going to expand so much. I would have to think how would be the easiest way and as well provide some compatibility of reading the old files.

lalo-uy, 2016-09-15:
I believe that adding the #Include option will be the easy way.

onekk, 2016-09-15:
I have done an approach that it was easy for a CAM program that I ve started to write but now is abandoned.

I have defined a python module that hold the global variable named glb.py and defined some empty dictionaries.

plus for every file i have wrote the rules  that check the esistence of the files and state two predefined location, defined in the init routine of the program based on the os identificator.

Linux hold the data in the .config hidden directory

Windows maybe in the $user directory i didn't remember well if it is in %APPDATA% or similar 

```
def check_tool_table(self):
    if glb.f_tooltable is None:
        if glb.localini == 1:
            EC_L.create_toolfile(self, "./EuroCAM")
        else:
            EC_L.create_toolfile(self, "~/.config/EuroCAM")
    else:
        pass

    read_tool_table(self)        
```

THis function read the tooltable and pupulate the 

```
def read_tool_table(self):
    '''
    Read the tools config file and populate the tools dictionary
    '''
    config = ConfigParser.SafeConfigParser()
    config.read(glb.f_tooltable)
    if config.sections() is not []:
        glb.Tools={}
        tdata = []
        for name in config.sections():
            for data in glb.Tooldata:
                tdata.append(config.get(name, data))
            if glb.debug[4] == 1:
                print "ec_logic:read_tool_table > tdata = {0} ".format(tdata)

            glb.Tools[str(name)] = tdata
            tdata = []

```

When necessary the dictionary is dumped to file with this routine and there is even an update routine that write the tooltable and reread if necessary.

```

def write_tool_table(self):
    config = ConfigParser.SafeConfigParser()

    for k,v in glb.Tools.iteritems():
        config.add_section(k)
        for i,d in enumerate(glb.Tooldata):
            if glb.debug[4] == 1:
                print "ec_logic:write_tool_table > k,v,str(i) = ", k,d,str(v[i])
            config.set(k,d,str(v[i]))
    with open(glb.f_tooltable, 'wb') as configfile:
        config.write(configfile)

def update_tool_table(self):
    write_tool_table(self)
    self.Log.append("Re Writing ToolTable")
    read_tool_table(self)
    self.Log.append("Re Read ToolTable")
```

with this approach every variable is defined with a name and it is more compact to address in the code

in the glb.py the 

```
def ini_search_paths(file_name):
    paths = map(
        lambda path: os.path.join(path, file_name),
        (
            '~/.config/EuroCAM/',
            './EuroCAM/'
        ),
    )
    for path in paths:
        if os.path.isfile(path):
            return path

# These variable are defined here to be used in all the modules.
# Some of them are translated in the main loop

inif_name = "eurocam.ini"
toolf_name = "tooltable.ini"
machf_name = "machtable.ini"
wp_name = "wptable.ini"

inifile = ini_search_paths(inif_name)
f_tooltable = ini_search_paths(toolf_name)
f_machtable = ini_search_paths(machf_name)
f_wptable = ini_search_paths(wp_name)
````
A variable hold the fields under each tool item:
````
Tooldata = ["sha","dia","rad","len","ovl","shd","flu","cc","opt"]

```

But maybe there is another better appoach.

In this manner the Dictionary is imported and "propagated" importing only the glb module (global variable module).

But I'm not a professional programmer.

Regards