vlachoudis/bCNC #425: Can not select communication-port

ErikHGV, 2016-09-10:
Hi 
I have been looking at bCNC for a while. It really looks good and seems to provide a lot of good features. So I'm now in the process of getting it to work, and this is causing me some problems.

I have installed Python 2.7.12 and pyserial 3.1.1 together with the bCNC 0.9.4 on a Windows10 PC.
The bCNC  is opening allright, but without communication to the grbl (Arduino Uno clone with USB).
Clicking on the communication shows only the Reset-icon (no selection of ports) and when clicking the reset, the message shows that it can not access /dev/ttyusb0.
In general the screens in bCNC looks pretty much like what you see in the documentation.

With the exact same connection I can access the grbl with the Universal GcodeSender (Java) and from  a terminal-program in Windows and it works fine with these.
When I try with the bCNC I have made sure no other program is using the same port (at least what I think!).

Also within the Python I can access the grbl with the wxTerminal (py-based), so the communication port is visible in the Python.

I need to say that I'm new to Python, so it may be a stupid mistake I have made.

Can any one lead me in the right direction to solve this?

Thank you very much!

ErikHGV, 2016-09-10:
Correction after new clean install:

The message showing when pressing the "Hard Reset" by the communication shows:
<<
Error opening serial

could not open port ": WindowsError(3, 'Den angivne sti blev ikke fundet.')
>>
[it says that the specified path was not found]

ErikHGV, 2016-09-10:
OK - after using my eyes a bit I found that communication is set in the "File"-tab.

Now I have connection - sorry for the disturbance!

closed 2016-09-10 by ErikHGV

Effer, 2016-09-12:
Hi @ErikHGV ,
I was waiting to test it in a Win10 environment but nice to here you found it. 
If you need further help let me know