vlachoudis/bCNC #428: [Info] It is possible to define ribbon on two lines?

onekk, 2016-09-11:
In my interface, due to muy screen resolution 1024x768 and i suppose more important with 800x600 resolution of some RPi, some ribbon elements, especially these in the Editor are too wide for the screen, it is possible to have two lines so it is possile to see all the button?.

Or maybe make the ribbon scrollable with an orizzontal scrollbar, two arrow buttons one at the leftmost  and one rightmost position of the ribbon space?

Thanks

robcop2000, 2016-09-12:
Same request if is possible. I have RPi with 800x600 display

Effer, 2016-09-12:
Some times ago it was 800x600 compliance, then we added too many stuff and forgot to check back compatibility. 
Time to refactor some stuff.

vlachoudis, 2016-09-15:
Indeed, several options are added and is no longer fitting. I believe the best approach will be to make the ribbon to automatically adjust for smaller environments. Like making the big-icons smaller, maybe removing the text-label from some icons or push the excess of icons in a drop-down box.