vlachoudis/bCNC #429: [Suggestion] Buttons in the Control

onekk, 2016-09-11:
For my daily operation i lack some buttons:

two buttons to switch on and a button to switch off the coolant/air/aspiration

some buttons to have a FF  at least in the four main direction.

Typical case is i have a 0.1 mm step for fine tuning

I need also 50 or 100 mm to move away from the area

so if it was possible to add below the main buttons the FF buttons, and make them selectable from the ini file it will be a great improvement, or maybe add a grid of user buttons (like the 0 button in th centre) above the main jogging buttons, it seems that there is some space left here.

Thanks and Regards

onekk, 2016-09-13:
Maybe some of the problem maybe solved with the file included in https://github.com/vlachoudis/bCNC/issues/431

At least for the coolant part,

closed 2016-09-13 by onekk

vlachoudis, 2016-09-15:
@onekk You have already the settings of 3 predefined steps in the keys 1,2,3 as step1=0.1, step2=1.0 and step3=10.0

onekk, 2016-09-15:
In fact maybe I've thought to put in the interface three buttons that recall the 3 pre settings in a quick manner, instead of the multiply x10 and divide x10, in my everyday use I use rapid movemente of 50 or 100 and a finer of 1  and a finest of 0.1mm almost esclusively on Z to find the zero of the workpiece.