vlachoudis/bCNC #430: CAM improvement Save dialog?

onekk, 2016-09-12:
@Effer has commited some code to save the GCode, but I didn't find how to save the file:

For a better CAM usage experience (similar to some CAM program) I can't get some beahviour,

In example, I want to save only the selected GCode and i want to save it uppercase.

I'm expecting to find a "save as.." with a button/checkbox/option  "only selected GCode" and "Uppercase".

Plus I'm resolving to save some "flatten plugin" generated code and then modifying it with an editor to add some GCode in front of the code (but after the comment header) "G90 G20 G17 G59" and then "M8 M3 S1200"

and at the end of file "M5 M9 M2" and some other stuff.

I see that there is some field in the ini file that could do the trick notably:

[CNC]

header = 
footer =

But I have some difficult to put a  newline in this fields when editing in the Machine listbox "Enter" simply close the editing window.

Plus a little cosmetic note, some Machine definitions maybe are better to be put near the "Stock" and "Tool" in the "database" under "GCode", I'm referring to the:

Units (inches)
Lasercutter
Start up
Spindle min (RPM)
Spindle max (RPM)
Header gcode
Footer gcode

That are relevant to the CAM operation (maybe the "Start up" is related to some $ field in GRBL?)

Only some thinking to improve bCNC, I'm trying to add some buttons to turn on  (M8) and off (M9) the coolant/vacuum but I'm havong some problem with the code.

Effer, 2016-09-12:
Hi @onekk
it's still in beta stage so there's no dialog for now. 
- If you want to test it, select the type **GCode clean** from the Save dialog 

[image]

Let me know if the output is correct.

- Try hit SHIFT+ENTER to add a new line in the multiline fields. For me it works in Windows.

onekk, 2016-09-13:
Ok tried now on the machine ( running Linux, Debian Jessie armhf version):

1) SHIFT + ENTER will do the job in inserting the correct newline in the fields

2) using the txt extension, save in Uppercase and without the inactive blocks.

closed 2016-09-15 by vlachoudis