vlachoudis/bCNC #432: dxf import

zozigizo, 2016-09-15:
I'm having problems importing DXF file (from freecad) into bCNC.

On command line I get message:
"Cannot handle entity type: INSERT in layer: 0"

If I create dxf file with librecad it works 

fine and can import into bCNC.

In error report popup I get:

`Program     : bCNC
Version     : 0.9.6
Last Change : 24 Aug 2016
Platform    : linux2
Python      : 2.7.12 (default, Jul  1 2016, 15:12:24) 
[GCC 5.4.0 20160609]
TkVersion   : 8.6
TclVersion  : 8.6

Traceback:
Traceback (most recent call last):
  File "/home/z/Downloads/bCNC/Utils.py", line 389, in __call__
    return self.func(*args)
  File "./bCNC.py", line 1871, in loadDialog
    if filename: self.load(filename)
  File "./bCNC.py", line 1932, in load
    Sender.load(self,filename)
  File "/home/z/Downloads/bCNC/Sender.py", line 388, in load
    self.gcode.importDXF(filename)
  File "/home/z/Downloads/bCNC/CNC.py", line 2358, in importDXF
    path.fromDxfLayer(dxf, entities, units)
  File "/home/z/Downloads/bCNC/lib/bpath.py", line 1121, in fromDxfLayer
    b = bulge[i]
IndexError: list index out of range`

I tried to upload the dxf file but it keeps failing when I try to upload... it's a very small file (1619 bytes)

zozigizo, 2016-09-15:
`  0
SECTION
  2
HEADER
  9
$ACADVER
  1
AC1009
  9
$INSBASE
 10
0.0
 20
0.0
 30
0.0
  9
$EXTMIN
 10
0.0
 20
0.0
 30
0.0
  9
$EXTMAX
 10
0.0
 20
0.0
 30
0.0
  0
ENDSEC
  0
SECTION
  2
TABLES
  0
TABLE
  2
VPORT
 70
0
  0
ENDTAB
  0
TABLE
  2
LTYPE
 70
1
  0
LTYPE
  2
CONTINUOUS
 70
0
  3
Solid line
 72
65
 73
0
 40
0.0
  0
ENDTAB
  0
TABLE
  2
LAYER
 70
1
  0
LAYER
  2
0
 70
64
 62
7
  6
continuous
  0
ENDTAB
  0
TABLE
  2
STYLE
 70
1
  0
STYLE
  2
STANDARD
 70
0
 40
0
 41
1.0
 50
0.0
 71
0
 42
1
 3
ARIAL.TTF
 4

  0
ENDTAB
  0
TABLE
  2
VIEW
 70
0
  0
ENDTAB
  0
ENDSEC
  0
SECTION
  2
BLOCKS
  0
BLOCK
  8
0
  2
FILLET
 70
0
 10
0.0
 20
0.0
 30
0.0
  3
FILLET
  0
LINE
  8
0
 62
0
 10
-39.5
 20
28.0
 30
0.0
 11
39.5
 21
28.0
 31
0.0
  0
ARC
  8
0
 62
0
 10
-39.5
 20
25.0
 30
0.0
 40
3.0
 50
90.0
 51
-180.0
  0
ARC
  8
0
 62
0
 10
39.5
 20
25.0
 30
0.0
 40
3.0
 50
-0.0
 51
90.0
  0
LINE
  8
0
 62
0
 10
-42.5
 20
-25.0
 30
0.0
 11
-42.5
 21
25.0
 31
0.0
  0
LINE
  8
0
 62
0
 10
42.5
 20
25.0
 30
0.0
 11
42.5
 21
-25.0
 31
0.0
  0
ARC
  8
0
 62
0
 10
-39.5
 20
-25.0
 30
0.0
 40
3.0
 50
180.0
 51
-90.0
  0
ARC
  8
0
 62
0
 10
39.5
 20
-25.0
 30
0.0
 40
3.0
 50
-90.0
 51
0.0
  0
LINE
  8
0
 62
0
 10
39.5
 20
-28.0
 30
0.0
 11
-39.5
 21
-28.0
 31
0.0
  0
CIRCLE
  8
0
 62
0
 10
19.0
 20
-24.5
 30
0.0
 40
1.4
  0
CIRCLE
  8
0
 62
0
 10
19.0
 20
24.5
 30
0.0
 40
1.4
  0
CIRCLE
  8
0
 62
0
 10
-39.0
 20
-24.5
 30
0.0
 40
1.4
  0
CIRCLE
  8
0
 62
0
 10
-39.0
 20
24.5
 30
0.0
 40
1.4
  0
ENDBLK
  0
ENDSEC
  0
SECTION
  2
ENTITIES
  0
INSERT
  2
FILLET
  8
0
 62
0
 10
0.0
 20
0.0
 30
0.0
  0
ENDSEC
  0
EOF
`

zozigizo, 2016-09-15:
I guess it's related to the FILLET in the dxf? I looked at bpath.py, there are entries for LINE, CIRCLE, ARC, POLYLINE, SPLINE and LWPOLYLINE but no FILLET?

zozigizo, 2016-09-15:
The only difference is the one that doesn't work has a circle added - A diff between the two files shows:

`130,132c130,132
< ENDSEC
<   0
< SECTION
---
> BLOCK
>   8
> 0
134c134,144
< ENTITIES
---
> SKETCH002
>  70
> 0
>  10
> 0.0
>  20
> 0.0
>  30
> 0.0
>   3
> SKETCH002
194a205,238
>   0
> CIRCLE
>   8
> 0
>  62
> 7
>  10
> -30.760513
>  20
> 19.186481
>  30
> 0.0
>  40
> 3.266176
>   0
> ENDBLK
>   0
> ENDSEC
>   0
> SECTION
>   2
> ENTITIES
>   0
> INSERT
>   2
> SKETCH002
>   8
> 0
>  10
> 0.0
>  20
> 0.0
>  30
> 0.0
`

Effer, 2016-09-15:
Hi @zozigizo,
can you zip the file and upload them here, probably git hub doesn't accept some file format?
Can you try to explode the component before saving. In that way it seem to work for me.

zozigizo, 2016-09-17:
Hi - I've tried several times to upload the dxf file - the country I'm currently in is restricting uploads it seems. 

The explode idea worked great - I couldn't find an explode function in freecad, but I opened the exported dxf file in librecad, exploded it, saved it and then it loaded into bCNC great.

Thanks for the tip!

closed 2016-09-17 by zozigizo

Effer, 2016-09-18:
No problem. It will help to have the original file to debug the problem. I take the text you posted and build it here: [importNotWorking.zip](https://github.com/vlachoudis/bCNC/files/478685/importNotWorking.zip)

As reference if someone would like to dig in the dxf import issue. It seems like the importer library doesn't like the entity [INSERT](http://www.autodesk.com/techpubs/autocad/acad2000/dxf/insert_dxf_06.htm)
That's seem to be simply a group of object (moved, roteted, etc). Here in the file is simply "named" as SKETCH002, but not being handled rise the exception.
As tested by @zozigizo exploded file (no group) avoid the problem.