vlachoudis/bCNC #437: Convert bCNC to exe Setup File without any library

MohamadAlsadi, 2016-09-23:
I need this program as exe file (installer) without need any library  on PC client .
I try to convert it exe file , I used two program for that(cx_Freeze and installer) but I have Error win I run exe file.  look to Attach picture
note : when I convert to exe I put the main Page is( bCNC.py) . look to this file for setup :
import sys
from cx_Freeze import setup, Executable

setup( name = "AnyName", version = "3.1", description = "Any Description you like", executables = [Executable("bCNC.py", base = "Win32GUI")])

what is problem ??? 

[image]

Effer, 2016-09-23:
Hi  @MohamadAlsadi ,

try this:
 in the bCNC.py change this line

```python
PRGPATH=os.path.abspath(os.path.dirname(__file__))
```

with this

```python
if getattr(sys, 'frozen', False):
	# frozen
	PRGPATH=os.path.abspath(os.path.dirname(sys.executable))
else:
	# unfrozen
	PRGPATH=os.path.abspath(os.path.dirname(__file__))
```

In this way I can build a really nice exe that seems to works, and could be added to the distribution.

Here my setup.py

```python
import sys
from cx_Freeze import setup, Executable

# Dependencies are automatically detected, but it might need fine tuning.
build_exe_options = {"packages": ["os"]}

# GUI applications require a different base on Windows (the default is for a
# console application).
base = None
if sys.platform == "win32":
    base = "Win32GUI"

setup(  name = "bCNC",
        version = "0.1",
        description = "Grbl sender and more",
        options = {"build_exe": build_exe_options},
        executables = [Executable("bCNC.py", base=base)])
```

and the I run **python setup.py build**
Copy  everything from the dist folder to the main folder to access graphical resources etc.

Keep us posted about your test.

MohamadAlsadi, 2016-09-23:
Hi I change every thing as you tell . but when I run exe file I get error message
[image]

MohamadAlsadi, 2016-09-23:
Hi @Effer 
can you send me the exe file on my email : 
ise34857@gmail.com

Effer, 2016-09-23:
Here it is @MohamadAlsadi ,
just unzip and run bCNC.exe
Tested on Win10 (with python and other stuff) and WinXp (clean machine).
With tested I mean that it start... can't try with the machine right now:
[bCNC4Win.zip](https://github.com/vlachoudis/bCNC/files/489582/bCNC4Win.zip)

MohamadAlsadi, 2016-09-23:
ok thank you @Effer 
it is work

Effer, 2016-09-23:
Wow, never hope this could work at first shot.  A ready exe without dependency to install will be a nice addition for windows users

vlachoudis, 2016-09-23:
@Effer Does it require python to be installed?

MohamadAlsadi, 2016-09-23:
No
it not required paython
I try it on windows 10 without paython

Effer, 2016-09-29:
Tested with a clean PC (Win10) from USB key.
Even the camera stuff work (opencv, PIL) but still some of plugin is not loaded. It seems like it can't recognise "from fractions import gcd" that I left on some of the plugin even if not used.

Anyway really a cool feature. @vlachoudis  what you think about add a branch with the exe for windows?
I could keep it in sync with the release if you dont' have access to a win machine.

vlachoudis, 2016-09-29:
@effer is a good idea. How big is the executable?

Terminator88, 2017-04-14:
Thank advanced to Effer :+1: . I search very much website to solve my issue, but unknow that I just copy everything from build folder to the main folder. Ha ha ha :)

Effer, 2017-04-14:
@Terminator88 you're welcome.

Terminator88, 2017-04-15:
@Effer I have new issue, I unknow why, file .exe still run in process manager when I exit software. Do you have any solve for this ?
[image]

Effer, 2017-04-15:
@Terminator88 never see that. Are you using cx_freeze or py2exe?

Terminator88, 2017-04-15:
I use cx_freeze like you, I don't know why your software can complete shutdown, but my software is still run underground.

hznupeter, 2018-01-22:
@Effer your exe is worked on my loptop. 
I hope the exe could be put into release

hznupeter, 2018-01-22:
the exe has a small bug,it can't be closed by close button or exit button

justinclift, 2018-05-07:
As a data point, since it sounds like there's a working .exe + a few support files pretty much ready to go, it probably wouldn't be very difficult to use something like [NSIS](https://sourceforge.net/projects/nsis/) to package it into a reasonable Windows .exe.  That'd give people a simple install/uninstall process with sane defaults (eg install location).

FatNarval, 2018-10-30:
Hi folks, it would be nice to see bcnc4win with new updated features! 
Any progress so far?

Harvie, 2018-10-30:
Also there's some project called  PyInstaller

https://medium.com/dreamcatcher-its-blog/making-an-stand-alone-executable-from-a-python-script-using-pyinstaller-d1df9170e263

FatNarval, 2018-10-30:
Thanks, I'll try my luck with this one!

Harvie, 2018-10-30:
@FatNarval thanks. but note that you first have to get bCNC working on your machine, because these packagers will most likely get the needed libraries from your computer.

Harvie, 2018-11-13:
I want to have this in 0.9.15 release. is there somebody to prepare the build process?

FatNarval, 2018-11-16:
I can not do it, anyone else?

Harvie, 2018-11-20:
There is now much easier way of installing bCNC on Windows:
https://github.com/vlachoudis/bCNC/issues/964#issuecomment-440315914

Harvie, 2018-11-21:
I've managed to build the .exe, you can try beta here:

https://github.com/vlachoudis/bCNC/releases/download/0.9.14.29.beta.exe/bCNC-0.9.14.29-beta-git-exe.zip

Build process seems to work well. You can expect similar build to be part of 0.9.15 release.
I am closing this for now. Please let me know if something is wrong. But note that this .exe is built on top of development bCNC version, which might have some problems unrelated to .exe build itself. So please first check that reported problem happens only in .exe version, but not when running in python directly.

FatNarval, 2018-12-01:
Camera feature seems not to work at all .

FatNarval, 2018-12-01:
In Windows 7, when bCNC gui window is minimized to taskbar, while active dialog box/window of bcnc is active, bCNC keeps minimizing itself, effectively preventing itself from using again, until restarting it.

Harvie, 2019-01-09:
I've just realized that we might not need windows builds at all. We can just compile some kind of loader to that .exe and load the .py files using it... So there will be no need to rebuild the loader as long as there will be no changes to loader itself or the requirements.