vlachoudis/bCNC #440: Suggestion: interface improvement

onekk, 2016-09-23:
Thanks for the great efforts in developing bCNC.

I have some suggestion on how to improve the interface.

1) visualize a grid on the workarea or even some cross spaced say every 50 mm so it is easy to move the gantry near a determined position?

2) Visualize maybe on demand with a toggle button, the WCS Origin points, so it's easy to spot the WCS origins

3) When the GCode file is running populate the relevant boxes on the "State" with the data on the gcode file, like  feed and speed present in the gcode being executes, (I know that it is not in sync with what is executed by grbl but it will help to spot some errors.)

4) Display a messagebox with the first lines of the Gcode file where generally you will find the comments, about the settings used to produce the gcode.

Fosforitofercho, 2016-12-27:
I have others, excuse me use your space:
1. Play, pause and stop buttons on the Probe tab. It is very important for errors like proobe not connected
2. Possibility of fixing the terminal all the time, on the right will be good
3.Go to Work zero by default

Thank you . I use bCNC 0.9.7

excuse me

vlachoudis, 2016-12-27:
No.3 from @onekk is now functional with grbl v1.0
@Fosforitofercho all what you mention can be done right now by customizing your .bCNC file. (documentation is missing though :)
 1. Add a line like
``` probe.ribbon = Connection Probe Run```
in your .bCNC file in the [bCNC] section
2. The same you can add the "Terminal" in which ever panel you like. Look the bCNC.ini file in the bCNC directory (but not modify this one)
3. Work Zero by default I don't understand this one. However you can do what ever you like in any of the user customizable buttons. Click on any of them and right your gcode there.

Effer, 2016-12-27:
The for point 1 and 2 is here https://github.com/vlachoudis/bCNC/wiki/Customize%20Menu
here other customization 
[image]

vlachoudis, 2016-12-27:
Thanks @Effer I missed that :)

Fosforitofercho, 2016-12-29:
ok I gona see the file .bCNC . and look what i can do ,¿where I can foind the documentation ??

The 3 point I found the button with this function , ups! :P

Another suggestion  , is posibble a Change tool alarm , when the program start this routine  it could warm with a sound , it will be great, because we dont need be togheter the machine all the time. 
Thank you very much for quick response  .
The program is amazing

vlachoudis, 2017-01-02:
@Fosforitofercho That could be an easy option to add, on every warning, error message to play a user defined sound.

closed 2017-01-02 by vlachoudis