vlachoudis/bCNC #443: TCP/IP connection to grbl is possible to do ?

levis0, 2016-09-26:
I wrote little program for ESP8266 to act as access point and listen to a specific port. After wireless connection to it with telnet i can control bcnc. Connection is transparent. Can be added to bcnc option TCP/IP:PORT connection option, not only physical serial interface ? It eliminates 1 wired connection and i think will be much more stable.

levis0, 2016-09-29:
Now i'm trying ESP-12 with firmware esp-link from jeelabs and virtual serial ports emulator from Eterlogic. Seems all is working.

vlachoudis, 2016-09-29:
I've tried once something like this using the Linux socat command. I don't remember how, but if you search on the web you can find examples how to bind a virtual serial to a file or a TCP/IP address

lalo-uy, 2016-09-30:
There are many options but this is ok

socat  pty,link=/dev/virtualcom0,raw  tcp:192.168.254.254:8080&

closed 2017-01-02 by vlachoudis