vlachoudis/bCNC #444: how do I use G code generated by HSMWorks?

robermoncam, 2016-09-27:
anyone knows how to use G code already generated by another program?

jeffmd, 2016-09-27:
Just use the file open icon/menu option to load a g-code file generated by another program.  I haven't used HSMWorks but it should not be a problem. You will have to select the grbl post processor in HSMWorks before exporting the nc file and then just load it in bCNC.

onekk, 2016-09-29:
As an information, I use all the time G-Code generated with CamBam and selecting the appropriate post-processor in this case Grbl it works, but take care of cecking the maximun length of the line as in older grbl version is limited to 50 characters, newer if i didn't go wrong is 80 characters another possible issue may come from unsupported codes notably G40 is not supported in some versions [see]( https://github.com/grbl/grbl/wiki)

vlachoudis, 2016-09-29:
@onekk just a correction, canned cycles are expanded and used inside bcnc

onekk, 2016-09-30:
Thanks @vlachoudis I have missed that point, What canned cycles are supported? eventually i will write a post processor for bCNC for CamBam that supports canned cycles.

closed 2016-10-03 by vlachoudis