vlachoudis/bCNC #448: Request: How to add "CAM" controls

onekk, 2016-10-01:
using bCNC i miss some simple "routines". 

Example, I have to cut a piece of wood and i have to go from say X0,Y0 to X120,Y0 and cutting that length in some z-pass.

Supose i have some buttons, [MIN] [MAX] [ZMIN] [ZMax] and [Zstep] [LINE] [Execute]

It will be easy to:

move the gantry to X0,Y0 and press [MIN] it memorize XO YO as minpos
move the gantry to X120,Y0 and Z to the upper stock surface and press [MAX] and [ZMAX]
move the gantry to wathever X or Y and Z = the lower surface of the stock (if you have table maybe the = of the table or the = of the supports) and press [ZMIN]

Select the "operation" say [ZSTEP] and in a windows that appear or elsewhere enter 1.5 
 [CUT] but maybe a more complex thing like [SQUARE] or [POCKET] and press [Execute]

bCNC calculate the necessary step and send to grbl the appropriate codes without emitting a GCode file.

When I try to implement this i found a big problem:

How to add a frame to the bCNC interface, say I want a "button frame" under [Control] with the operational buttons described above.

It will be feasible with a plugin but having a bunch (8 or 10) buttons will be more intuitive and resemble some "Hardware Controller" found in production machine (I haven't used a production machine, but i have seen some people using them positioning the gantry in contact wit the piece, pressing some buttons, and cut the piece without generating a line of GCode)

Sorry for my bad thinking and worse explanation.

Regards

pitzerwm, 2016-10-02:
I agree totally, anything that makes your work easier is a good thing.

vlachoudis, 2016-11-21:
@onekk falls in the same spirit on what @Effer proposed some time ago to have a dialog than just a list for the plugins, then all this functionality could work.
I will try to think on how to integrate some custom frames and buttons for the plugin in an easy way to interact with the program