vlachoudis/bCNC #452: Webcam/Camera configuration for RPi3

CCSnell, 2016-10-11:
I have a raspberry pi 3, running on Protoneer's image version 3.00. PIL, ImageTk,and OpenCV all setup (import in IDLE returns no errors). The camera,  v2.1 is connected via the ribbon cable and the icon in bCNC is no longer greyed out. Using the following command shows the camera functioning correctly

`$ raspistill -o output.jpg`

But I cannot get the camera to show on the pendant view or in the program. I believe the problem is in the .BCNC or in BCNC.INI setup. In the wiki it says to manually edit .BCNC to enable the camera. I cannot figure out what parameters need changing or what to change them too. Any guidance would be appreciated.

vlachoudis, 2016-10-12:
Does the opencv recognizes the rpi camera?
You can try in python the following
```$ python```
```import cv2```
```camera = cv2.VideoCapture(0)```
```camera.read()```

CCSnell, 2016-10-12:
Running the following gives no errors:

```
$ python
import cv2
camera = cv2.VideoCapture(0)

```
Running the following gives me a live stream: 

```
# import the necessary packages
from picamera.array import PiRGBArray
from picamera import PiCamera
import time
import cv2
 
# initialize the camera and grab a reference to the raw camera capture
camera = PiCamera()
camera.resolution = (640, 480)
camera.framerate = 32
rawCapture = PiRGBArray(camera, size=(640, 480))
 
# allow the camera to warmup
time.sleep(0.1)
 
# capture frames from the camera
for frame in camera.capture_continuous(rawCapture, format="bgr", use_video_port=True):
	# grab the raw NumPy array representing the image, then initialize the timestamp
	# and occupied/unoccupied text
	image = frame.array
 
	# show the frame
	cv2.imshow("Frame", image)
	key = cv2.waitKey(1) & 0xFF
 
	# clear the stream in preparation for the next frame
	rawCapture.truncate(0)
 
	# if the `q` key was pressed, break from the loop
	if key == ord("q"):
		break

```

Still not working in bCNC.

nbernard, 2016-10-15:
I have a similar issue: in my case, it is with a Raspberry Pi 3 too, but with a raspbian image. OpenCV and the python bindings are the one from Debian, installed using the packaging system.

A USB camera (UVC) works as expected (image in bCNC) when plugged on the Raspberry Pi, but the rpi camera (v1.3 NoIR in my case) doesn't (while it works with raspistill/raspivid for instance).

DaveDischord, 2016-10-19:
I'm experiencing the same issue, with raspistill working fine.  The Camera in bCNC is still greyed out, even with a USB camera plugged in.

danst0, 2016-12-07:
Similar issue here, with my Logitech Quickcam Pro. 
The test commands you mentioned work, but there is no picture in pendant.

Daniel

```
$ python
import cv2
camera = cv2.VideoCapture(0)

camera.read()
```

vlachoudis, 2017-01-02:
@DaveDischord do you have the python-imaging-tk pakage installed?
@CCSnell apparently the default opencv doesn't recognize the RPi camera

closed 2017-01-02 by vlachoudis

HCS-II, 2017-02-16:
Thanks for the great GRBL UI!

Seems a Python interface package called `picamera` can properly interface the RPi camera with OpenCV. See: [https://github.com/waveform80/picamera](https://github.com/waveform80/picamera).

nbernard, 2017-02-25:
Hi,

I modified Camera.py to be able to use the Raspberry Pi's camera. A preliminary version is here: https://github.com/nbernard/bCNC/tree/rpi_cam
Its pretty much a hack, but I guess it could be useful until OpenCV has native support for these cameras.

You need to put 'aligncam = -1' in the '[Camera]' section of your .bCNC file, as I'm using negative indices to distinguish this case from cameras handled directly by OpenCV.

I would be glad if some people were able to test this (with rpis but with other plateforms too, to be sure this introduce no regression with other cameras) before I create a pull request for merging into the main repository.
To me, it seems slower than with a UVC camera. If you have suggestions to make it faster, they are welcome.

vlachoudis, 2017-02-26:
@nbernard that's great.

sjamesparsonsjr, 2018-05-10:
Hello @nbernard, let me see if I'm getting this right.  First, I update my Camera.py from the rpi_cam tree.  Then add 'aligncam = -1' to the '[Camera]' of .bCNC above the 'aligncam_d = 0.0'.  I've done this and it seems that bCNC still won't recognize the RPI Camera.  Should I fork the whole repo?

sjamesparsonsjr, 2018-05-10:
Its works once you reboot :D