vlachoudis/bCNC #453: Probe Question

pitzerwm, 2016-10-16:
Being a noob, I know that I am missing something simple, but I have been reading and watching for days and I can't find what I think I am looking for.

I made a plate and alligator clip, its wired into the probe  terminals on the BOB.  In Bcnc, I see the probe command and after placing the bit slightly above the plate, it lowers to the plate and touching it, stops.

What I am missing is where do I add/input the pate thickness?  I also have no idea what the other blanks need, but one thing at a time.

Thanks for the help.

pitzerwm, 2016-10-17:
No answer???

atrueresistance, 2016-10-17:
Hello @pitzerwm  I believe you may be looking for the [tool change](https://github.com/vlachoudis/bCNC/wiki/Tool-Change) functionality. 

I can't be much help in setting the tool change part up, as I just manually jog the machine over and probe a known area and z+- to where I need it and record the offset. Then when I switch tools I z probe the same spot as before and move z +- to the offset I wrote down. 

I haven't tried the tool change tool for a long time > 1 year, but when I initially tried it never paused even after I had indicated pause before and after. I'm sure that is fixed, but stand by your machine and be ready to hit stop just in case.

pitzerwm, 2016-10-17:
I don't think this is a tool change issue, but would be used if you did change the tool.  When you set up the Z0 it would appear that you use a plate and lower the bit until it touches.  But there is the thickness of the plate to take into consideration.  If you just use manual gcode, you state it.  This looks like it should be automatic.

Here is a picture of that page, I thinking that somewhere you put that thickness measurement.

The only place that makes sense to me is the TLO, but then I'm a noob, so could be way off base.

[image]

Effer, 2016-10-18:
Hi@pitzerwm, maybe I'm wrong but the probe command is only for touch the stock material in bCNC. It make the gantry move to desired xy position and then it goes down with z until it reach your metal piece. When triggered it stop and report back the quote. No need to input any thickness. From there you can zero your WCS and work on the surface of it.
If you need to autolevel a surface you have to move to the another tab.
What do you need to do with the probe?

pitzerwm, 2016-10-18:
Hi Effer,  Watching other utubes etc, it appears that you need to tell the machine where the bit is in reference to the work, so it knows the start of the depth that it is going to cut.  Some people just use a piece of paper and call it even.  When you use the probe command, it goes down until it touches the plate (working with wood) and it stops, but the thickness of the plate needs to be taken into consideration, as its usually higher than .001mm of the piece of paper.  In my case the plate is 1.61mm, so someplace you need to let the machi9ne know that you are actually above the surface that height.  In bcnc it appears to be automatic, and someplace you just tell it the thickness.  With all of the wizards on here, I'm surprised someone hasn't told me.  As you can see there are a few blanks on the form and obviously they are there to make your positioning easier.  In the manual, that tab is just skipped, or at least I can't find it.

I understand that the autoleveling is different, haven't needed to go there yet, but there is more info in the manual on that tab.

Thanks for trying to help.

Effer, 2016-10-18:
So, to summarize your procedure:
- Put a stock o material (wood) on your work area
- Place the gantry over the piece
- Put a metal plate (1.61mm) over the stock and clips and alligato
- Probe 

From here I'll go on in this way:
- Put the plate height in the z Wpos
[image]

- Hit Z=0 so the value is saved for Z coordinate for the current workspace
- Rise Z and remove the plate
- Run the job (it should keep the 0 as wood surface if the thickness is correct)

Can't try on machine just now and I have a mechanical probe so usually my way is different.

pitzerwm, 2016-10-18:
Thanks Effer!!  I knew that it wasn't rocket science, just have to hold your mouth right :o) AND put the right amount in the right place.  Did you mean to put 1.61 in the z Wpos slot?

Effer, 2016-10-18:
Yes. And then hit the "Z=0" button.

pitzerwm, 2016-10-18:
Just a thought, wouldn't you actually need -1.61?

pitzerwm, 2016-10-21:
I found that you just enter the thickness of the plate and DO NOT hit Z=0

closed 2016-11-05 by vlachoudis