vlachoudis/bCNC #456: Connected does not change

atrueresistance, 2016-10-29:
It seems there is an issue with the latest master. I connect and the menu does not change, nor does the XYZ coordinates render. 

[image]

Clicking on it seems to work pop up this. 
[image]

Now even though this renders incorrectly, I am able to set my WCS and run files.

vlachoudis, 2016-10-31:
Could you check on the "File" as well the "Terminal" tab if it detected as Grbl V1 or GrblV0.9

atrueresistance, 2016-10-31:
I'm running `CarbideMotion 0.9g` as of current.

auhopu, 2016-10-31:
I had a similar issue after the last pull. The status showed: "Not connected".

I could pass commands to my controller (e.g. home, jogging), the controller would execute the commands,  but there was no response back in bCNC. The DRO was empty and the spindle location was not shown on the diagram.

My GRBL is 1.0c and my File/Serial/Controller was set at "Grbl" (as opposed to "Grbl-V0").

With reverting to the previous state, the issue is corrected.

vlachoudis, 2016-10-31:
Ok, I will downgrade my grbl and give it a test

jeffmd, 2016-10-31:
I am using 0.9j GRBL and have had no problems with running bCNC git master.

vlachoudis, 2016-10-31:
Ahh. Did CarbideMotion changed the init string from "Grbl 0.9g" to "CarbideMotion 0.9g"?

chamnit, 2016-10-31:
@vlachoudis : I think they did a while back, if I remember right. I think they changed it back to `Grbl v0.9g` to address compatibility. Some users may have `CarbideMotion`.

atrueresistance, 2016-10-31:
@chamnit I'm not sure the firmware was what shipped with the Rev 3 board from Carbide3d. It hasn't been flashed at all.

vlachoudis, 2016-10-31:
@atrueresistance @auhopu could you check if with the new modification that should detect the string CarbideMotion it works?

auhopu, 2016-10-31:
I just tried the latest pull, but still get the "Not connected" status and no DRO. Homing & jogging ok.

Also, my previous remark of seeing no response back in bCNC is partly true. $$, $, $G, $I work in terminal. However "?" just returns "ok", no state or current position (my $10=3 ). My GRBL version is 1.0c.20151109 .

Reverting to the previous version works normally.

vlachoudis, 2016-11-01:
@auhopu Can you tell me what is the init string returned by Grbl after reset?

auhopu, 2016-11-01:
This one:

Grbl 1.0c ['$' for help]

atrueresistance, 2016-11-01:
@vlachoudis Lastest changes to sender.py solve my problem. Numbers and connected status are back for my machine.

closed 2016-11-21 by vlachoudis