vlachoudis/bCNC #459: Positioning problem of the cutter Run/Stop

Mr-Cozi, 2016-11-21:
I have a problem positioning the cutter when I start machining, when I press RUN!

And when I stop the Gcode the tool resumes its position ...
Watch the red circle on the attached screenshot, the position of the tool is the same between the two images ...

What is the problem ???

[image]

[image]

Guenni75, 2016-11-23:
I have a similar Problem.
If i Stop a GCode, often the Z Value is changing.

Mr-Cozi, 2016-11-23:
Do you find solution ?

If you watch my screenshot, only MPOS Z remains the same between a run and a stop.....

vlachoudis, 2016-11-25:
@Mr-Cozi @Guenni75 I didn't understand the problem?

Mr-Cozi, 2016-11-26:
My cutter stay at the same place but the software see différent position... 
See the red spot between the two screenshot!

Mr-Cozi, 2016-12-04:
ok, i solve my problem...
I checked my g-code program and at the beginning i found the command G92 X... Y... Z...,

I deleted this command and know it's ok !

vlachoudis, 2016-12-04:
Thanks for reporting @Mr-Cozi I am closing the issue

closed 2016-12-04 by vlachoudis