vlachoudis/bCNC #460: Probing outside bar to find center axis

MarkDerbyshire, 2016-11-24:
Hi

I don't know if this has been suggested or implemented or is doable but is there a way to find the center of a horizontal bar using probing.  Like center circle but on the outside of a round (or square) bar.  My thought process would be to set y axis to a point on the circumference probe then the Z axis.  Lift to safe height move Y axis to other side of bar and then probe the Z axis again until it touches.  Take the readings of the Y and Z points and calculate the center of the bar axis if there is such a calculation :)

--
Mark

vlachoudis, 2016-11-25:
@MarkDerbyshire no there is not such a functionality built-in. However one can program it and set it in the user programmable buttons, using the function evaluation capabilities. I believe in one of the old issues it was reported how to make the center probing, so you could follow a similar logic for your task.

MarkDerbyshire, 2016-11-27:
Thanks.  I shall give that try.  For now I have just used the command box using G38.2 code on the Y axis to probe one side, set Y to zero then probe the other side and then divide the distance by 2 to give me center.  I can do the same with the X axiz to find the face center of a round bar (or square for that matter )

--
Mark

closed 2017-01-02 by vlachoudis