vlachoudis/bCNC #461: Tool change in inches?

RandyG-G, 2016-11-28:
First of all, thank you Vasilis for this very useful program.  I am using it with grbl 1.1 development on my Nomad 883 desktop mill.

I have all the settings entered, and can run single-tool programs fine.  I like the interface and the jogging and am eager to try auto-leveling when I can add a probe socket in parallel with the tool probe.

I have all the settings in bCNC and grbl in millimeters, inheriting from the settings in Carbide Motion.

However, I do all my CAD in inches.  For setting up a workpiece I convert zero offsets etc. to mm for setting the DRO values and that works fine.

I cannot do toolchanges successfully.  I touch the first tool on the top of the workpiece, zero the Z axis, and click Calibrate which takes the tool up and over to the tool probe, measures it, and updates the Calibration window.

But when I run the gcode and it hits the first M06, the tool goes up and over to the tool sensor, runs downwards rapidly and tries to push the tool "down through the desktop".

This is very repeatable and I am quick with the e-stop button but I will not try it again.

Since my gcode is in inches, the first line is G20.  The M06 lines (my typical program has a roughing and finishing cutter, and sometimes one or more drills) come after this.  Can the toolchange internal macro be confused by the inch setting or have I failed to set some parameter correctly in bCNC?

As I said, the workaround is to separate the gcode into single-tool programs, but since you have provided the neat tool-change capability I would like to take advantage of it.

Thanks,

Randy

vlachoudis, 2016-11-29:
@RandyG-G I will try to reproduce it on my machine. Maybe I forgot somewhere the metric->imperial conversion.

RandyG-G, 2016-11-29:
Thank you.  I made a simple gcode that only has 

G20
M06 T1
G0 X1 Y1 Z1
M06 T2
G0 X1 Y1 Z1
M06 T3
G0 X1 Y1 Z1

After I zero the tool on my wasteboard and calibrate (which goes well) I run the gcode.

I can verify that the distances and speeds during tool change are in inches at the first M06 tool change.

Also, the G0 line causes the carriage to move in the wrong direction and limit-error.  The Z DRO value also is wrecked.

If I were to set bcnc and grbl to operate strictly in inches, is there a list or guide somewhere of all the settings I need to convert?  I have read that grbl is always in millimeters internally so I don't know what of the $$ values I would need to change.

Randy

vlachoudis, 2016-11-29:
For bCNC you have to go to "Tools -> Config" and tick the "Units (inches)"
and for grbl you have to set $13=1.
On the new bCNC you can do it in "Tools -> Controller"
I will try to merge the Config and the Controller so bCNC to read (is already done) and use the config of grbl.

RandyG-G, 2016-11-29:
Thank you again, @vlachoudis .  Do I also need to convert $11, $12, $24 to $132 from mm to inches, or does the $13=1 handle that?

Randy

chamnit, 2016-11-29:
@RandyG-G : All Grbl settings are in mm. $13 only changes Grbl feedback units, including reports, g-code state, and parameters. It does not alter the g-code state.

RandyG-G, 2016-11-29:
OK, thank you, Sonny.  That will save my fingers some calculator-pushing.  So the sometimes-inches and sometimes-mm is totatlly the responsibility of the controller program (i.e. bCNC, Carbide Create, etc)?

Randy

RandyG-G, 2016-12-01:
I will close this issue.  With $13=1 and $N0=G20 in grbl and inches set in bCNC, tool changes occur properly in an inch gcode file.  I think that my not having G20 set in the grbl startup block is what was causing the problem.

closed 2016-12-01 by RandyG-G