vlachoudis/bCNC #463: bCNC Freezes/Crashes at 97-99% completion

RGUERI1, 2016-12-08:
This seems to be something new, it just started happening a couple weeks ago. I didn't run my machine for a few months, but I don't remember making any changes. My configuration is the following:

Smoothieboard 5X
Windows 10
Python 2.7
bCNC 9.4, 9.6, 9.7
CamBam

In short, anytime I run a gcode file, the progress bar stops between 96-99% and the program essentially freezes. I can hit stop and it goes into alarm state. After that, I cannot get any movement, by jogging or gcode file, without reconnecting or hard resesetting. I have tried using xx.nc and xx.gcode files. I have watched the terminal and the last code block, m5 says ok and m30 says ok. However, it never seems to finish. I also use Laserweb 3 for laser cutting and the same file runs perfectly and completes. 
I've tried re-installing Python 2.7, to no avail. I've tried using M02, instead of M30. Nothing seems to work.
Any ideas? Can someone point me in the right direction, to correcting this problem? What's baffling is that nothing, to my knowledge, has changed since the last time I used it and I've done many successful cuts with no problem.
I can provide screen shots or log files, if needed.

vlachoudis, 2016-12-08:
Could you share your gcode so I could check?

RGUERI1, 2016-12-09:
G21 G90 G64 G40
G0 Z3.0
T0 M6
G17
M3 S0.3
G0 X5.4035 Y6.0779
G0 Z0.5
G1 F300.0 Z-0.001
G1 F900.0 X7.2824 Y11.8787
G1 X5.4035 Y6.0779
G1 X3.5168 Y11.8787
G1 X5.4035 Y6.0779
G0 Z3.0
G0 X6.8398 Y6.0
G0 Z0.5
G1 F300.0 Z-0.001
G1 F900.0 X7.9746 Y8.877
G1 X9.1152 Y6.0
G1 X7.9746 Y8.877
G1 X6.8398 Y6.0
G0 Z3.0
G0 X7.2383 Y7.0273
G0 Z0.5
G1 F300.0 Z-0.001
G1 F900.0 X8.707
G1 X7.2383
G0 Z3.0
M5
M30

RGUERI1, 2016-12-09:
This is actually one of the laser files, hence the "S0.8." I have a Dewalt Router, no VFD Spindle.

RGUERI1, 2016-12-09:
here's another through cambam, setup as a milling/cnc:

( Made using CamBam - http://www.cambam.co.uk )
( FLEXITEST-10 TAG MASTER 12/6/2016 7:13:12 PM )
( T0 : 0.2 )
G21 G90 G64 G40
G0 Z3.0
( T0 : 0.2 )
T0 M6
( Engrave3 )
G17
M3 S1000
G0 X12.0924 Y42.4309
G0 Z0.5
G1 F300.0 Z-0.2
G1 F800.0 X14.2115 Y38.7199
G1 X12.0924 Y42.4309
G1 X14.2115 Y45.9123
G1 X12.0924 Y42.4309
G1 X9.9537 Y45.9123
G1 X12.0924 Y42.4309
G1 X9.9537 Y38.7199
G1 X12.0924 Y42.4309
G0 Z3.0
G0 X17.0045 Y38.7297
G0 Z0.5
G1 F300.0 Z-0.2
G1 F800.0 Y45.9221
G1 X18.274
G2 X18.9576 Y45.8147 I0.0081 J-2.1772
G2 X19.5289 Y45.4924 I-0.493 J-1.5414
G2 X19.9195 Y44.9407 I-1.0321 J-1.1448
G2 X20.0709 Y44.1301 I-1.8144 J-0.7582
G2 X19.8658 Y43.3098 I-1.6025 J-0.0352
G2 X19.3775 Y42.7287 I-1.8261 J1.0388
G2 X18.7867 Y42.3821 I-1.3764 J1.669
G2 X18.274 Y42.2698 I-0.5859 J1.4482
G1 X17.0045
G1 X18.274
G3 X18.7867 Y42.3821 I-0.0731 J1.5602
G3 X19.3775 Y42.7287 I-0.7857 J2.0157
G3 X19.8658 Y43.3098 I-1.3378 J1.6199
G3 X20.0709 Y44.1301 I-1.3973 J0.7851
G3 X19.9195 Y44.9407 I-1.9656 J0.0523
G3 X19.5289 Y45.4924 I-1.4226 J-0.593
G3 X18.9576 Y45.8147 I-1.0643 J-1.2191
G3 X18.274 Y45.9221 I-0.6755 J-2.0702
G1 X17.0045
G1 Y38.7297
G0 Z3.0
G0 X20.0904 Y38.7199
G0 Z0.5
G1 F300.0 Z-0.2
G1 F800.0 X17.7418 Y42.2502
G1 X20.0904 Y38.7199
G0 Z3.0
M5
M30

RGUERI1, 2016-12-09:
Attached is a new/clean output from CamBam. I've tried so many adjustments
and modified the others so many times, that I sort of lost track. This was
the original attempt, to engrave, that is getting me stuck.
Thanks Rustin

RGUERI1, 2016-12-09:
A light just went off as I typed the last reply. This may be the first time
that I tried to use the "engrave" function in CamBam, I normally use
"profile" or "pocket." Could this be part of my problem? I'll test it in a
little while and make sure I'm not crazy!

RGUERI1, 2016-12-09:
I figured it out. It was actually my X-axis Max Limit Switch! It was
apparently stuck between open and close. I noticed that I would get an
alarm as soon as I started the print. After clearing it, the cut would
continue and then hang up on the 98% range. Funny how it only affected bCNC
and not Laserweb?? I'm not sure, but it's working. Thanks Guys.

ruenahcmohr, 2016-12-10:
just a note that I'v never had bCNC finish a file from bamcam, it always hangs on the last (or after) line
the job finishes, but the sender gets stuck. 
The only downside is that I cant tell how long the job took cause its still counting time.

latest grbl on uno at 9600, no autolevel, windows xp, nc file from cambam

jkoljo, 2016-12-11:
I am also seeing this issue with Smoothieboard 3X. I am using it to control a PCB milling machine. My G-code ends with :

M5
M2

And that works fine, the spindle stops and Smoothie seems happy with it, but bCNC gets stuck at last percent or so. If I stop gcode execution at bCNC, Smoothie goes to alarm state and it may or may not recover using bCNC's reset functions.

vlachoudis, 2017-02-05:
@RGUERI1 the G64 generates an error:20 from grbl.

J-Dunn, 2017-06-07:
@vlachoudis  
GRBL does not support G64, and v1.1 returns an error code,  so what is the conclusion to this one? 

Is this bCNC needing to update to handle the newly added error codes or is it lacking any error trapping? Getting stuck does not seem very good and your comment identifies the problem does not say what you think needs to be done and this does not have bug status. 

OP could just remove the unsupported Gcode but this has highlighted a lack of error trapping and handling of the GRBL return codes. These returns were added to improve the interface with senders, wouldn't it be good to take advantage of them?

vlachoudis, 2017-06-07:
@J-Dunn this was discussed in length in the past. The correct approach is to tune your gcode generator to create something that is compatible with your controller (grbl in this case), otherwise bCNC should stop and report the error returned by grbl. Now if for some x/y reason you want to ignore those codes you can instruct bCNC to do so, by adding in the [Error] section of the .bCNC ini file the gcode and the behaviour like
g64=1

closed 2017-06-07 by vlachoudis

J-Dunn, 2017-06-07:
I agree the correct solution to OP's problem is to remove the unsupported Gcode ( or use the handy config option) , but from the context of bCNC, I would suggest that either it does make use of the GRBL return , or if that is not worth supporting, that it deals with unexpected returns in a better way than getting stuck. 

That is what error trapping means. 

There was a reason for the unexpected return from the controller and that can be fixed but the next time one of the several supported controllers returns something other than what is expected, you will have another untrapped error and your sender will be stuck again.

vlachoudis, 2017-06-08:
@J-Dunn so if I understand well your proposal would be when an error happens, instead of simply stopping the g-code execution to execute instead some predefined user code. Could be a nice addition, simply I am wondering if this might generate more problems than what it is supposed to solve. The only think I would imagine I would like to do in such a case is to perform a software reset.

J-Dunn, 2017-06-08:
Well not really. This bug is titled "bCNC Freezes/Crashes at 97-99% completion". There was a reason for that but it does not seem to be well handled if the bug description  is accurate and bCNC is freezing. It seems that there was no indication of the error and the only way out is hard reset of the controlled device. 

https://github.com/gnea/grbl/wiki/Grbl-v1.1-Interface

> error:X: Something went wrong! Grbl did not recognize the command and did not execute anything inside that message.

If I understand this correctly GRBL is ignoring the line and it is bCNC which is choosing to send the next Gcode.

My way to look at this would be to say if GRBL is prepared to continue, bCNC should probably be prepared to continue sending. The error msg "error 20" should be visible to the user ( which he appeared to report was not the case. It seems bCNC "freezes" before getting to the point where it reports the output coming back from board, which was what caused it to halt. 

In this case it seems that it could have continued and GRBL would have safely ignored it. In a case where it does cause problems, that is for the user to diagnose from the hopefully complete output returned from the board and logged by the sender.

There could be a 'halt on error' option if you think some users would prefer that, although halting a job and having to reset loses position and means restarting or scrapping the piece. 

Going into freehold may be a better option since this could be recoverable and does not mean losing position.  So there are a number of ways to handle this but it seems that there is currently a bug in that bCNC detects that an error code was returned and halts execution ( possibly unnecessarily )  and forgets to report to the user the return msg which caused it to stop sending. Not good. 

> I have watched the terminal and the last code block, m5 says ok and m30 says ok. 

The very minimum would be for the terminal to show "error 20" as was returned on the serial connection and some message from bCNC explaining that is had stopped sending. 

I hope that makes it by suggestion clearer.