vlachoudis/bCNC #464: On the fly feed rate adjustment

danst0, 2016-12-09:
Hi,

I had this issue now already a few times, having problems to pinpoint the cause.
Following setting: Load one GCODE file, reduce the feed rate to 70%, cut the file.
Leave the setting at 70% (unchanged), load another GCODE. The issue now comes when I cut the second file, the actual feed rate now is even higher than the setting in GCODE.
Worse, even functions like probing seem to go with insane speeds.

Can anyone confirm this behaviour?

Daniel

Effer, 2016-12-10:
Hi @danst0 ,
I believe it happened even to me one time but I couldn't figure out how to reproduce it.
Thanks for feedback, now I can have a deeper look.

vlachoudis, 2016-12-11:
@danst0 which version of you have grbl v0.9 or v1.+?

danst0, 2016-12-12:
0.9. Thanks for the hint, I didn't even know that there was a 1.0...:)

vlachoudis, 2016-12-15:
In V1.+ the feed rate is synchronized constantly according to what grbl is reporting versus what the user has set. For v0.9 the Fxxx feed rate commands are corrected only when they are found in the commands to send, which looking at the code right now is not reset after the program stop/start and needs to be fixed.

Effer, 2016-12-15:
Good spot @vlachoudis , I gave a brief look but didn't find the point. I'm far from machine now and can't do any test. 
FYI in windows there's some problem to reset the override to default value. With double click some time it quite work (setting to 99% or 101%) but right click is not processed at all. Maybe add a reset button will be more clear and compatible.

Fosforitofercho, 2016-12-29:
Similar problem, but when starts the gcode, if i put the indicator for example 25% , next I start the g code sending , it runs at 100% feed speed , the way is move the indicator , in this moment starts the override .   Is more logic start the program  at fraction of feed selected in the indicator  and adjust it on the way , for me is the right manner , not start in 100% , even the indicator indicates other value .

I use bCNC 0.9.7 and grbl v1.1

Thank you

vlachoudis, 2017-01-02:
@Fosforitofercho thanks I see the problem. I will correct it in the next respin.

vlachoudis, 2017-01-02:
The correction is already on the latest master

closed 2017-01-02 by vlachoudis