vlachoudis/bCNC #465: Object not shown in BCNC

danst0, 2016-12-10:
Hi,

I just stumbled across an issue where stuff in the GCODE file is not shown in the BCNC interface.

Please have a look at the result and the pictures. The part of the Dino head is missing in the interface, it is however cut.

I also attached the DXF and Code.

Daniel

[image]
[image]
[Archiv.zip](https://github.com/vlachoudis/bCNC/files/644055/Archiv.zip)

Effer, 2016-12-10:
Hi @danst0 ,
please check if increasing the timeout rendering time, the issue persist.
This time out is set with the small combobox in the top right corner of the canvas. In your screenshoot is set to 5 (seconds).

vlachoudis, 2016-12-22:
Any news on that?

danst0, 2016-12-22:
Sorry for the delay. Increasing the value mitigated the problem. However it's not really obvious...🤖

vlachoudis, 2016-12-23:
Python is not the fastest for drawing, gcode can be really large, and tkinter can only draw from the main thread. So we had to find a balance between what to show and when to unlock the interface.

closed 2016-12-23 by vlachoudis

danst0, 2016-12-23:
Limit rendering makes total sense then!
Couldn't there be a window popping up, asking for permission to continue/warn that some items might not be displayed?

Daniel

vlachoudis, 2016-12-23:
@danst0 That could be an option. But it has to be smart and shown only the first time for each file otherwise it will be annoying on every redraw.

poesel, 2017-01-19:
I'd like to second that motion. I just came here to report it as a bug. Some notification would be nice.

vlachoudis, 2017-01-19:
@poesel there is a notification on the status bar, most probably nobody pays attention since is not a popup :(

Effer, 2017-01-19:
What about change the color label in the Toolbar?
(this one)
[image]

Or add an **exclamation point** in the lower corner of the canvas.

Fosforitofercho, 2017-02-02:
i had never seen that , because doesn’t have tag like time out, change this value to greater number fix the problem and render full Gcode :O  , thanks Effer . Each time I convince me that bCNC is the perfect software for GRBL
[image]