vlachoudis/bCNC #466: Probing ignores feedrate when feed override is set

tuxintrouble, 2016-12-16:
If a feed override is enabled, the probing function ignores the probing feedrate. This usually results in the tool crashing into the Tool length sensor. If Feed override is reset / disabled, the probing funktion uses the probing feedrate again.

vlachoudis, 2016-12-18:
It's a good point. We should protect it.

vlachoudis, 2017-01-02:
Now that I was looking the code, I think the easiest would be to automatically reset all feed overrides when probing is invoked.

tuxintrouble, 2017-01-02:
That would be cool for me. 
Without knowing how this is handled in detail, I was wondering if this isn't actually rather a GRBL issue. Some sort of protection through bCNC might be good, anyways.