vlachoudis/bCNC #468: bCNC does not open first time

bvernham, 2016-12-21:
The first time I run the BAT file bCNC does not open.  Python does.  If I drag the bCNC.py file into the opened python window then it opens.

After that, if I close bCNC, it will reopen by just opening the bat file.

I am using the Anaconda2 distribution package at it was the only way to find all the necessary windows 64 bit extensions.

Ideas?

Thanks

Bruce

vlachoudis, 2016-12-27:
I cannot help much here, since I don't use Windows :)

vlachoudis, 2017-01-28:
Did you find any solution? I am closing the issue for the moment

closed 2017-01-28 by vlachoudis

CharlesEBM, 2018-11-04:
A bit late maybe, but I had the same problems.  It was all down to the numpy module Not being installed.
When I installed it ("pip install numpy") bCNC opened with no problems

Harvie, 2018-11-04:
@CharlesEBM your solution is right, however this ticket was created in 2016 and closed in 2017, while dependency on numpy was introduced in 2018. so it's not related.