vlachoudis/bCNC #471: Islands?

vederstein, 2016-12-26:
I'm very impressed in your software and congrats to you.

I do have a feature enhancement request though (along with the multitude of others out there):

It would be very advantageous for the addition of the creating of "islands."  Many 2-1/2 and 3D cam systems have this.  The operator chooses an outer and inner boundary.  The pocket is the area between the boundaries.

For my use it would be to mill a index pin onto a flange plate.  The pin would then go into the hole on another plate assuring concentrically of the two components.

Thank you for your time.

vlachoudis, 2016-12-27:
Its something on my TODO list, however recently I don't have enough time to work on that.

Harvie, 2018-07-02:
duplicate of #220

closed 2018-10-29 by Harvie