vlachoudis/bCNC #473: Manual change tool Funtion ( Using the button for test)

Fosforitofercho, 2016-12-27:
When i use the funtion CHANGE ( tab Probe- Tool - button Change  ) at the end step ,when machine returns to initial position , moves on  X, Y and Z at the same time, the correct is : first X and Y, and after Z .  However when i make a  milling and T0 M6 command launch the tool change, works correctly  X and Y first and final Z .  I know that is a easy fix, but i dont know modificate the program .  I use bCNC  0.9.7  . Thank you so much

vlachoudis, 2016-12-27:
I don't understand why it should do that, since it calls exactly the same function CNC.toolChange, that is called whether is manually invoked or from the gcode. In the toolChange function the restoring is performed as```
		# restore state
		lines.append("g90")		# restore mode
		lines.append("g0 x[_x] y[_y]")	# ... x,y position
		lines.append("g0 z[_z]")	# ... z position
```
where you can see it executes the g0 in two steps first the X&Y as you mention and then followed by the Z motion. Can you verify on the "Terminal" if the commands are properly send to grbl?

closed 2016-12-27 by vlachoudis

Fosforitofercho, 2016-12-29:
Ok i will try again , if happens the same i'll write to you. thanks for answer