vlachoudis/bCNC #474: bCNC RPI3 Alamode CNC Shield

DenCNC, 2016-12-27:
I am trying to get bCNC to run on  my Raspberry Pi 3 with Alamode and CNC Shield v3.  I am getting a few errors at startup that will not allow me to connect to the serial port.  I have installed python pyserial module several times and no success.  Here is the first error:

[image]

and then I get this error when I try to select the correct serial port:

[image]

I have tried everything that I can think of.  I seem so close that I hate to give up.  Any thoughts?

DenCNC, 2016-12-28:
Never mind. Removed pyserial-3.2.1 and installed pyserial-2.7 and everything works perfect now. Thanks.

vlachoudis, 2017-01-02:
@DenCNC thanks for letting us know. I will try to investigate why pyserial-3.2.1 is failing.

vlachoudis, 2017-01-29:
@DenCNC I've tried today the pyserial-3.2.1 and it works ok on my pc

closed 2017-01-29 by vlachoudis