vlachoudis/bCNC #475: Autolevel Color doesnt work

Fosforitofercho, 2016-12-29:
I installed the pillow package with    comand  "easy_install Pillow" from CMD ,

After i open the bCNC, make autolevel but i cant see the elevation as a colour map , 
doesnt works.

Someone can make it works on windows ??   or can gime me help ?

Thank you so much 

I use bCNC 0.9.7

vlachoudis, 2017-01-02:
Could you try to see if ImageTK is present, open python and type
```
from PIL import Image
from PIL import ImageTk
```

Fosforitofercho, 2017-01-15:
Ok, i typed that and nothing message appeared , after i reopen bcn ,load a .nc code and .probe file, but again i only see numbers :S

vlachoudis, 2017-02-05:
@Fosforitofercho Do you have numpy installed?

Fosforitofercho, 2017-02-19:
Man my mistake, I didn't know the existence about   automatic installers of python libraries, i had installed Pillow using the console, after ,i found a windows installer of PIL (http://www.pythonware.com/products/pil/)  , click, run and Voilá!  the colors of autolevel works, thank you , escuse me the inconvenience . Yes I Now, sometimes I make trivial questions , but I am not an expert and I never has programmed on python .

vlachoudis, 2017-02-20:
If you want, you can document your findings in the wiki page to help others

closed 2017-02-20 by vlachoudis