vlachoudis/bCNC #478: auto re-execute file number of time

MohamadAlsadi, 2017-01-01:
I need to added textbox in bCNC to re-execute the Gcode file, number of times equal to the value in textbox automatically without user intervention . because I need cut the same shape number of time . 
please help me to add this future

vlachoudis, 2017-01-07:
I don't like adding too many specialized buttons on the display. It is already cluttered. The easiest option to implement it would be do add an argument in the command line like
RUN <times>
and it will push <times> the gcode as is in the stack.
Of course at this point it cannot make any discrimination so it will add the header/footer as well.

Effer, 2017-01-07:
--run is already an arg option

MohamadAlsadi, 2017-01-07:
@Effer 
I don't understand what you say @Effer  please give me more explain?

Effer, 2017-01-07:
@MohamadAlsadi there has been already a user that asked for a similar request time ago. So there's a line argument that can run bCNC on start. Combine this with the name of the file to load you can run the file directly.
Anyway this will not repeat the job as you request.
I'm looking if using python inside gcode this could be accomplished but seems there are some problem with **%wait** and **%if running** special command.
@vlachoudis  **%if running** seems to not compile correctly the command. It substring the wrong number of char. Am I wrong?

MohamadAlsadi, 2017-01-07:
my dear please please please all story is : coy body G code and paste it multi number in the same pales 
how I can do that??
@vlachoudis 
@Effer

electrokean, 2017-01-07:
@MohamadAlsadi why don't you just use a text editor to copy & paste your G-code body into a new file along with the multiple work offsets?

Effer, 2017-01-07:
This is just a workaround: I've take the tiles plugin by @vlachoudis and adapted to your needs. You can use this as a base to satisfy you very special needs.

I can't upload python files here so you have to take the file copies.txt here and rename it to copies.py. Then move the file to the plugins directory in bCNC folder.

[copies.txt](https://github.com/vlachoudis/bCNC/files/691562/copies.txt)

Start o restart bCNC and you'll find a new function in the tools-->cam menu -->Copies

[image]

It's very basic just input the number of time the code must be copied (repeat) and then hit execute.
Hope this will help you to find a way.

MohamadAlsadi, 2017-01-07:
@Effer 
thank my dear but I need it in Edit tap not in tools tab. Please

MohamadAlsadi, 2017-01-07:
@Effer 
my friend
every block must start with line "g0 x0 y0" and end wit two lines "g4 P0.01 " and line "g10 l20 p1 y0"
for example: for 3 repeat

header
body
g0 x0 y0
body
g4 P0.01
g10 l20 p1 y0
g0 x0 y0
body
g4 P0.01
g10 l20 p1 y0
g0 x0 y0
body
g4 P0.01
g10 l20 p1 y0
footer

MohamadAlsadi, 2017-01-07:
@Effer 
no problem if the button in tools page. but every copy from body most start in line "g0 x0 y0" and end in two line "g4 P0.01" then line "g10 l20 p1 y0" 
for example: for 3 repeat

header
body

g0 x0 y0               // start first copy
body
g4 P0.01               //
g10 l20 p1 y0      //End first copy

g0 x0 y0              // start second copy
body
g4 P0.01               // 
g10 l20 p1 y0       //End second copy

g0 x0 y0              // start third copy
body
g4 P0.01   //
g10 l20 p1 y0   //End third copy

footer

Effer, 2017-01-08:
[copies.txt](https://github.com/vlachoudis/bCNC/files/691995/copies.txt)

If you need to edit the header or footer in the plugin you can do it from bCNC. To make a new line in the textbox use SHIFT+CTRL+ENTER

[image]

MohamadAlsadi, 2017-01-08:
@Effer 
Thank you my friend

vlachoudis, 2017-01-10:
I like the solution of @effer, anybody can make custom plugins to their needs and not inflate the basic code.

Effer, 2017-01-10:
Thanks @vlachoudis . Do you have a look to %wait and %if running? Do you believe it could be possible to run again the code with those commands append to gcode?

vlachoudis, 2017-01-11:
@Effer what is the problem with %wait and %if running? Anyway it wont be possible as it is now to re-run the code, since these commands do not support branching. We should put some kind of flow control in the program, to allow loops, function definitions etc. However it would be a mess to link it with the editor and the canvas for the display, since it will lose the 1-to-1 relation of a line and item to draw.

Effer, 2017-01-11:
@vlachoudis I take this opportunity to have some clarification. 
About this request my first thought was to solve it using some python inside gcode. So I supposed that adding an **%wait** (to empty the grbl buffer) and a **%if running initRun()** at the bottom will make the trick. I was wrong. 
1) How **%wait** works? With all the following lines or only with the command in the same line?
2) Am I wrong or if i write some code in the same line of **%if running** it will not be compiled correctly by **return compile(line[1:],"","exec")**  in the compileLine() method?

I never deeply thought the consequence of mixing gcode and python in the canvas rendering. It's really a mess. It's also a sin have python code ready integrated and can't use it at full power but just for  some very specific uses.

vlachoudis, 2017-01-12:
@MohamadAlsadi actually there is no need to make an additional plug in, since the "Tile" can do the job. If one specifies dx=0 and dy=0, ny=1 he can use the nx as the number of repetitions to do.

@Effer 
1. **%wait**, pauses the internal sender until all commands up to this point are completed, grbl return "ok" and the Rx grbl buffer is empty. There are some specific cases that is mandatory, e.g. in the probing we want to execute the python code in sync, after the probe command has finished. Otherwise bCNC will execute the following lines when they are pushed to grbl, which will yield wrong results since the probe variables wont be updated yet.
2. why you say that?
Indeed mixing the gcode and python in canvas is a mess. The program for every line of gcode keeps a pointer to the drawn object for only the gcode lines.
I don't have any clean and easy way of using the full power of python here, which would be nice.
As the program is now, it is a set of gcode lines with multiple inserts of python. The use all power of the scripting it has to be done the other way around, one python program with gcode lines inside. In this case we could render everything as a single object and all the editing/CAM capabilities will be gone.

MohamadAlsadi, 2017-01-12:
@vlachoudis 
thank you my friend . I wish that  if your answer comes early .
however thank you so much

closed 2017-01-12 by MohamadAlsadi

Effer, 2017-01-12:
@vlachoudis about **%if running**:

```python

		# execute literally the line after the first character
		if line[0]=='%':
			# special command
			pat = AUXPAT.match(line.strip())
			if pat:
				cmd  = pat.group(1)
				args = pat.group(2)
			else:
				cmd  = None
				args = None
			if cmd=="%wait":
				return (WAIT,)
			elif cmd=="%msg":
				if not args: args = None
				return (MSG, args)
			elif cmd=="%update":
				return (UPDATE, args)
			elif line.startswith("%if running") and not CNC.vars["running"]:
				# ignore if running lines when not running
				return None
			else:
				try:
					return compile(line[1:],"","exec")
				except:
					# FIXME show the error!!!!
					return None

```

The problem here is that the **line** that start with **%if running** is not executed when bCNC is not running (that's correct) but when running it'is compiled(and executed as python) starting from char 1 instead of char 11. If you put a print in the try block: **print line[1:]** you'll get it. 
Sorry I know I'm not really clear here.