vlachoudis/bCNC #483: Problem with autolevel

petsch75, 2017-01-07:
first of all, great, great program. I used it now a couple of weeks and use more and more functions. This week I started to use the autolevel function because I planned to mill a pcb. When I tried autolevel I got some strange results. 
First: bCNC stoped autolevel after one or two steps without any error message (Run completed 0%). Even no error from GRBL in terminal. I can probe at this points without any problem. And probe works before and after.  After a system restart it might happen that autolevel completed its work as expected.

Second: when I used autolevel data I got worse results. At the beginning everthing works fine, but as long as the job runs, I got deeper and deeper trenches (in the end through the whole pcb). It is independent of the size of the G-code file (I test diff. sizes), but every time the beginning is OK.

I think there is also something with probe. Whenn at the CONTROL panel feedrate is e.g. 120% and you do a "probe" the maschine runs with high speed down "Z" and even seems to ignore the probe-signal/contact. So far I expected that the feed rate in the PROBE panel rules.

Any idea why autolevel does not work with my system or did I something wrong? (Ubuntu 15.10/16.04)

vlachoudis, 2017-01-10:
Hi @petsch75, which version of grbl you are using?
1. Anyway, from this information I cannot understand the problem why it is stopping at mid-run. I never experienced, however I didn't happen to use the autolevel lately after I've upgraded to grbl v1.0.
2. The autolevel of course it is in-depended of the size of the data. bCNC is altering the code before it is send to grbl to follow the autolevel elevation information. Could it be a problem of acceleration or your gantry is heavy and when it goes up it loses steps, so with time it drifts downwards?
3. The probing feed should be done with what is entered in the field. However in a past version, if there was a Feed override it was applied resulting in a nonsense feed rate. In the current master it should reset the feed override to 100% before any probe command. If not then I have made something wrong.

petsch75, 2017-01-10:
@vlachoudis  thank you for your reply.
My autolevel test setups were:
GRBL 0.9j - Ubuntu Mate 15.10 (32bit) - bCNC 0.96 and 0.97
GRBL 0.9j - Ubuntu 16.04.1 (64bit) - bCNC 0.96 and 0.97
Gerber files ~ 5k and 450k size, PCB areas 25mm*25mm and 45mm*55mm
Probe line ist filtered (100n an CPU pin) and a twisted cabel (old ethernet patch-cabel)
Spindle ist totally disconected from mains (2 pole relay)

2. Well, the gantry or losing steps should not be the problem, because I milled the PCB (in are very unprofessional way) by using only probe. I centered the machine in the middle of the PCB "probed 0Z" and then milled the PCB. (I added some µm to be sure that no Copper is left, despite losing some µm width of the traces - 0.2 20° graving bit). 
Second I did some tests with my selfmade maschine and up to 1200mm/m for XY worked. So, for normal use I reduced it to max 1000 for XY and 500 for Z. Normally I am milling with 600XY und 100Z.

With the milled PCB I just updated hardware and GRBL to 1.1e. 

When I got new basematerial I will try autolevel + milling again. (A couple of PCBs are waiting)

I produced the Gerber data with KiCAD + Flatcam 8.5. Could this be the reason for autolevel mismatch?  Maybe the Gerber format does not fit with bCNC?

petsch75, 2017-01-10:
Sorry hit the wron button

petsch75, 2017-01-10:
Sorry again - I should go to bed now..

vlachoudis, 2017-02-05:
Hi @petsch75 could you save the autolevel information after the probing (extension .probe) and send it over to me together with your gcode.

petsch75, 2017-02-06:
Hi Vasilis,

sure I can do but unfortunately I can't do this before weekend.

Which software combinations you want me to test?
I have bCNC 0.96 0.97 0.98 and GRBL V1.1e on the system. If you would 
like to have GRBL 0.9j I need to downgrade before.

Regards Peter

vlachoudis, 2017-02-06:
I would prefer the latest grbl with the latest bcnc.

petsch75, 2017-02-07:
OK I will do.

By the way - last week I found an unusual reaction of the scan function.
I just milled a simple triangle. Than, to use all my material on the 
machine, I flipped the data and moved the origin with the internal bCNC 
editor. Afterwards the scan function get's confused. It moved two times 
on the x-axis, then startet to follow the outline (as normal) and 
finally stopped somewhere the part(area).

Try to reproduce it sometimes.

Regards Peter

petsch75, 2017-02-12:
Hello Vasilis,

attached is a ZIP file with all information in.
Setup and process is described in readme.txt

Greetings Peter

vlachoudis, 2017-02-12:
I don't see any attachment. Does github copies somewhere the attachments?

petsch75, 2017-02-13:
may be that github filtered it (or my provider ?).
In my send folder it is with attachment
I attched it again with this mail (1.1MB)

Effer, 2017-02-13:
You should attach the file via web form here on github issue page. 
Unfortunately attaching the file by mail in your client doesn't work.

petsch75, 2017-02-15:
Next try ... now its uploading already about 40minutes     
 what the hell is going on here...
[Uploading PCBtest.zip…]()
[Uploading Reame-Please.txt…]()

petsch75, 2017-02-15:
"Something went wrong we can't process this file "
A simple text file is not possible???
Sorry I have nor idear - drag and drop does not work, selcting a file also not possible - so how?
I will try it tomorrow again..

Effer, 2017-02-15:
Zip the file then drag and drop. Sometimes github is a bit hostile.

Kunzendorf, 2017-03-07:
Im experiencing problems with autoleveling too. 
Arduino Uno Clone with grbl0.9j firmware. 
I also tried with grbl 1.1j firmware, same results.  

G-code:
    G38.3Z-5F10
    G0Z2
    G0X12.5Y0
    G38.3Z-5F10 Nothing happens when this command is send.  

After first probing, it lifts of, move to next test point, nothing more happens.
I tried running the samme g-code commands manually from the Arduino Idé, without any problems.  

Im suspecting that the problem is someting with the buffer/data stream from bCNC to the Arduino, and i would like to give it some kind of delay between each probing, but im not sure where to do it in the program. 

Any help/suggestions  would be appreciated.

Mikael

lalo-uy, 2017-03-07:
There is a know bug in the clones that use the CH340G USB to serial
converter chip.
No walk arrounde.
Test on a board with ftdi or atmel chip

Kunzendorf, 2017-03-07:
Thanks for the fast reply.

I can mill without any problems, it is only when i do autoleveling there is a problem.   

This actually have a Atmel chip, but it would still properly be worth trying with an original Arduino board to make sure it is not what is coursing the problem.

petsch75, 2017-03-07:
The problem with broken communication is a problem I have too.

I am using a good quality and short cable, but it still happens from 
time to time (mostly in the middle of the job - of course).

There is no difference between my Arduino Uno clone or my original 
Arduino Nano (about half a year old - I got the feeling it's even worse 
than with my clone). I heard that there should be an firmware update for 
the Arduinos, but I don't know how to find out where to test or find the 
software.

Kunzendorf, 2017-03-08:
I tried with an original Arduino Uno today, same results as before, stops before probing second coordinate. 

I tried running bCNC from both my macbook and an Ubuntu operated system, exactly the same result.

Kunzendorf, 2017-03-08:
Solve the problem in my case.

It was the $1=25 setting in the grbl configuration, stepper motor disable time between  g-code commands, it was set to default of 25mS. 
When changing it to $1=255,  steppers motors  always enabled, and there is no problem when running auto leveling.

dguerizec, 2017-03-21:
Hello,
I found an oddity in [Sender.py](https://github.com/vlachoudis/bCNC/blob/master/Sender.py#L1079) that could explain the curious behavior described in @petsch75's [comment](https://github.com/vlachoudis/bCNC/issues/483#issue-199386966).

Shouldn't it read:
`CNC.vars["prbz"]-CNC.vars["wcoz"])`

vlachoudis, 2017-03-22:
@dguerizec many thanks for spotting it out, this was a bad one indeed. A typo on the grbl v1 branch only.
I've uploaded the patch, together with the possibility to save the gcode, expanded with autolevel information.

closed 2018-10-29 by Harvie