vlachoudis/bCNC #485: bCNC stops at Gcode line 4000

Ozzy62, 2017-01-09:
I have a strange issue with latest bCNC, it stops at gCode line 4000.
An older version (seems to be from 13/3-2016) does not have this problem.
Anyone have an idea as to what I'm doing wrong ??

Effer, 2017-01-09:
Can you post the full gcode?

Ozzy62, 2017-01-09:
Yes here you are.
The problem is that in the preview of the board more or less one 1/3 of the board was not seen.
I looked at the Gcode in a Gcode viewer and all was ok. I tried changing to and from metric and inches but that was not the problem. I then followed the stream of code until it stopped showing up on the preview. Loaded the file in programmers notepad, searched for the coordinates and that was line 4000.
It may be a coincidence, but then again.
 
[Hiland-Lab-PSU5.bot.etch.zip](https://github.com/vlachoudis/bCNC/files/694639/Hiland-Lab-PSU5.bot.etch.zip)

Effer, 2017-01-09:
The gcode takes some time to render so maybe it time out and doesn't draw all the job. You can adjust the timeout with the small combobox in the topright corner of the canvas:
[image]

This doesn't effect the streaming to the controller, it's just a way to make the GUI more reactive.
I run your code till line 6000 without problem. I've decrease the rendering time out to just 1 second and let bCNC stream data not visualized without stops.

When stop do you get some message from bCNC any report from the terminal?

Ozzy62, 2017-01-10:
Hi Effer
You nailed it. Changing the drowdown to 30 makes everything visible.
For the future, it would have very nice to have been warned, that the thing had timed out. 
And no, there are nothing in the terminal.
Thank you for your time.

vlachoudis, 2017-01-10:
@Ozzy62 this is only for drawing, since python is slow. Actually there is a warning message displayed on the status line that the rendering is too long and it is interrupted. I've added a Timeout label in the canvas to be more visible which setting to change.

closed 2017-01-10 by vlachoudis