vlachoudis/bCNC #487: Errore senza spiegazione

miomaox, 2017-01-10:
Bcnc funziona correttamente per AUTOLEVELLING ; SCAN DEL PROBE ; SPOSTAMENTI DEL CONTROLLER ; praticamente tutto OK, ma quando vado a incidere il contour di un PCB, lo inizia correttamente ma verso la metà del lavoro si inceppa e non risponde più a nessun comando ad eccezione dello spegnimento. A volte funziona lo stop, allora porto a zero nuovamente il mandrino e inizio una nuova incisione del contour, tuttavia, anche in questo caso il lavoro non viene mai portato a termine.
Per il GCODE io uso questo programma online: http://rapid-pcb.com/
Ovviamente ho bisogno di aiuto!
Grazie a chiunque possa prendere atto di questa mia richiesta.
1000 Cordialità
miomaox

miomaox, 2017-01-10:
English translation

Bcnc work correctly for AUTOLEVELLING; SCAN THE PROBE; MOVING THE CONTROLLER; virtually all OK, but when I go to affect the contour of a PCB, it starts correctly, but towards the middle of the work is jammed and not responding to any commands except shutdown. Sometimes it works the stop, then the port to zero again the mandrel and start a new recording of the contour, however, also in this case the job is never completed. For Gcode I use this program online: http://rapid-pcb.com/ Obviously I need help! Thanks to anyone who can take note of my request. 
1000 Friendliness 
miomaox

Effer, 2017-01-10:
Please post here the gcode that stop bCNC. You can zip and upload it here.

miomaox, 2017-01-11:
[bottom_contour x F80.zip](https://github.com/vlachoudis/bCNC/files/699653/bottom_contour.x.F80.zip)

Grazie per l''aiuto
miomaox

Effer, 2017-01-11:
I've full run the code without issue. The only strange thing I can see in the code is that there are some negative coordinate. The first at line 3474 where Y Axis to -0.008. Maybe this rise an allarm in GRBL?

You can try to move all the GCode a little away from origin. In bCNC you can do easly by selecting all the code (CTRL+A) and then input this in the Command textbox: **move 1,1,0** . This will move all the code 1unit away for both X and Y axis

(Ho eseguito tutto il programma gcode senza nessun problema. L'unica cosa strana che ho visto sono alcune coordinate negative per la X e Y proprio verso la metà del codice (riga 3474 la prima). Forse questo ferma GRBL?

Potresti provare a muovere l'origine del codice in modo da non avere problemi con i limiti di lavoro. Per farlo in bCNC puoi usare il comando muovi dell'interfaccia oppure selezionare tutto il codice (CTRL+A) e poi immettere il comando **move 1,1,0** nella casella di testo dei comandi (in basso a sinistra). Questo sposterà la lavorazione di una unità sia in X che in Y e non dovresti avere più coordinate negative.

miomaox, 2017-01-11:
Innanzitutto devo ringraziarti per la velocità nel rispondere.
Provo a seguire le tue ritte, poi vedremo cosa salta fuori.
La menata più odiosa è che quando succede il fattaccio, il programma si congela e non risponde a nessun comando se non quello dell'uscita: bottone rosso.
Quindi niente recupero delle coordinate, in pratica si può partire solo da zero, ma è una gara dura ritrovare il punto di partenza di prima del misfatto.
Quanto alla mia maldestrria nell'aprire e chiudere i messaggi, penso sia dovuto a un galoppante rincoglionimento senile.
Alla prossima
ancora grazie e ciao
Maurizio
  ----- Original Message ----- 
  From: Filippo 
  To: vlachoudis/bCNC 
  Cc: miomaox ; Author 
  Sent: Wednesday, January 11, 2017 9:05 PM
  Subject: Re: [vlachoudis/bCNC] Errore senza spiegazione (#487)

  I've full run the code without issue. The only strange thing I can see in the code is that there are some negative coordinate. The first at line 3474 where Y Axis to -0.008. Maybe this rise an allarm in GRBL?

  You can try to move all the GCode a little away from origin. In bCNC you can do easly by selecting all the code (CTRL+A) and then input this in the Command textbox: move 1,1,0 . This will move all the code 1unit away for both X and Y axis

  (Ho eseguito tutto il programma gcode senza nessun problema. L'unica cosa strana che ho visto sono alcune coordinate negative per la X e Y proprio verso la metà del codice (riga 3474 la prima). Forse questo ferma GRBL?

  Potresti provare a muovere l'origine del codice in modo da non avere problemi con i limiti di lavoro. Per farlo in bCNC puoi usare il comando muovi dell'interfaccia oppure selezionare tutto il codice (CTRL+A) e poi immettere il comando move 1,1,0 nella casella di testo dei comandi (in basso a sinistra). Questo sposterà la lavorazione di una unità sia in X che in Y e non dovresti avere più coordinate negative.

vlachoudis, 2017-01-12:
@effer Negative coordinates are not a problem. All my PCBs are in the negative space.

closed 2017-01-12 by vlachoudis

Effer, 2017-01-12:
I'm using the machine without zero. But if you reference an axis with a limit switch and then input a negative coordinate doesnt this could trigger the hardware limit switch?

vlachoudis, 2017-01-12:
The zeroing is for the absolute coordinates/machine the relative/work coordinates can be anywhere. And the default space is on negative coords. You remember the re-ocuring question why the space is on the negative side.

Effer, 2017-01-12:
My feeling about @miomaox issue is that he could zeroed the machine (Absolute Coordinate) but is using WCS in the same place. So if the part program call for a negative coordinate the machine travel back and hit the hardware limit switch. 
I have no doubt that bCNC and GRBL can manage negative coordinates.
I also agree that the problem is not bCNC related.

onekk, 2017-01-13:
If the program stop every time at a precise gcode "line" maybe it is a problem with gcode, but if the stop are random, probably the problem is related to the board not responding or some USB noise issue.

My two cents.

Regards

miomaox, 2017-01-26:
SOLVED !!!
Thank onekk. What I think the problem was due to USB noise.
I reinstalled Windows XP and now All operations are bcnc Excellent. No other trouble.
Thanks to all who have written to me to give me some help.