vlachoudis/bCNC #489: Overrides reset

Effer, 2017-01-11:
In the latest version Override reset button has been substituted  by a ComboBox to select which override affect. To be clear, this:

[image]

Now to reset you have to Right Click or Double Click the slideBar. This seems not to work correctly in windows. Right click doesn't work at all. While double click has the little drawback that it catch also a single click (if you do it in the dark gray slide bar), and so the override go back to 99% or 101%, depends if you do it at right or left of the slider.
I know the space is precious but could we add a little reset button? Maybe by horizontal splitting the combo box and use the top part to insert a small button. ( The scroll bar is already quite double size other buttons..)

vlachoudis, 2017-01-12:
Yes that could work.

vlachoudis, 2017-01-12:
I've just added it, thanks

closed 2017-01-12 by vlachoudis