vlachoudis/bCNC #490: lines in white when opening a dxf lile

dodolbysuround, 2017-01-11:
Hi everybody,
With the version v0.9.4, When I open a DXF, all the lines display in white, white, which make it difficult to work with, while in v0.9.1, the lines displays normally in black.
Any idea to help me solving this problem?

Effer, 2017-01-11:
Hi @dodolbysuround , it could be you save the DXF with white color? Color was introduced on 0.9.4 21 Apr 2016. We could add an option to customize the background color of the canvas.

dodolbysuround, 2017-01-11:
Well done, Thank you!

vlachoudis, 2017-01-12:
@dodolbysuround 0.9.4 is a bit old, this issue was corrected some time ago, where the white lines are changed back to black.
To change the canvas background color is a good option, but we have now to check not against the white lines but equal to background and change it to the negative one.

closed 2017-01-12 by vlachoudis