vlachoudis/bCNC #493: update 1.1e

robcop2000, 2017-01-15:
I have tryed to update to GRBL 1.1e. 
My configuration is Arduino Uno- GAUPS-Stepper DRV 8825.
GRBL 1.1.e- bCNC 0.9.7 as controller
As soon as I have update the 0.9.6 bcnc doesn't connect to the arduino.
I have updated but I had to delete the ./bcnc configuration  becuse the controller doesn't open. Or better start opening but at a certain point it closes automatically.
If I try to modiicate the new one, but doesn't work.
Tahnks
Roberto

robcop2000, 2017-01-15:
Another request is possible to download older version?

vlachoudis, 2017-01-15:
I've just downloaded the new grbl 1.1e and it connects without any problem. Normally it should autodetect the grbl, but for certainty please set in the "File" tab the "Controller" to **"Grbl"**.
"Grbl-V0" is for V0.9 or older.

You could download either from github in tab "Releases" or you could clone the git repository and then checkout any of the older versions.

Fosforitofercho, 2017-01-15:
+1 In my case works without problem with grbl 1.1e  , In conect menu ,  I select "Grbl", push conect  and works great

vlachoudis, 2017-01-29:
I am closing the issue, @robcop2000 if you can be more precise on what is the problem we can reopen it

closed 2017-01-29 by vlachoudis