vlachoudis/bCNC #496: "FIX" Spindle is not restarted after  change manually tool  while is running a G code

Fosforitofercho, 2017-01-16:
Good afternoon
 
When i run a g code, and M6 code appears  , the spindle is  tunr off , ok, afer i change normally the tool and claibrate it, final  i press restart spindle,  i wait the spindle restart by grb , but doesnt happens.

For it  i scrutinize the file "CNC.py" and add the following lines  in change manually tool  section 

# restore state
		lines.append("m3")      # restart spindle           (NEW LINE)
		lines.append("g4 p4")	# wait a  5 sec for starting of spindle   (NEW LINE)
		lines.append("g90")		# restore mode
		lines.append("g0 x[_x] y[_y]")	# ... x,y position
		lines.append("g0 z[_z]")	# ... z position
		lines.append("f[feed]")		# ... feed
		lines.append("[spindle]")	# ... spindle

After i tested it, and WORKS perfect !!

vlachoudis, 2017-01-28:
You should not hard code the M3 you don't always need a spinning spindle after a calibration. The last line in the code sends the [spindle] variable which contains the Mstate of spindle before the tool change started.  Now if it was working before, I have to understand why it didn't remember it.

vlachoudis, 2017-01-28:
@Fosforitofercho there was a bug in the evaluate function when there was a single line returning a string was not send to the machine. Now it is corrected.
Thanks

closed 2017-01-28 by vlachoudis