vlachoudis/bCNC #497: Sometimes the keyboard jogging  stops working

Fosforitofercho, 2017-01-16:
Hi ! :)

Normally after an error  or alarm  the keyboard control stops working,  obviously after an  error or alarm, i unloked the machine , however the keyboard jogging control doesnt works again and i have to move the machine with the virtual arrows (I Hate that !!!, excuse me the emotion XD, but is very stressful ) . Normally I have to restart the program for recover the full functionality .

How I can unlocked the control, or where i need to search for fix it ??

Thank you so much

jeffmd, 2017-01-16:
This might help:
https://github.com/vlachoudis/bCNC/wiki/Jogging

For me the ESC key does not help with the latest bCNC.  I have to move the mouse cursor over the Canvas area and left click.  You will notice a black outline of the canvas area when it has input focus and keyboard control will work.  Using the editor or some of the tools will cause the Canvas to lose input focus and keyboard control will not work.

Fosforitofercho, 2017-01-17:
It Works, is strange ,but it re enables the keyboard

vlachoudis, 2017-01-17:
@jeffmd is correct. The jogging with the keyboard only works when the Canvas has the focus. Otherwise if the focus is to anyother widget e.g. command line or editor, then the arrows have another function.
Esc should work always unless if it is intercepted by the widget which has the focus.

closed 2017-01-17 by vlachoudis