vlachoudis/bCNC #498: Failed to start

smooker, 2017-01-20:
[smooker@smoolap ~/src/bCNC]$ ./bCNC.py 
Traceback (most recent call last):
  File "./bCNC.py", line 2498, in <module>
    application = Application(tk)
  File "./bCNC.py", line 201, in __init__
    page.addRibbonGroup(n)
  File "/home/smooker/src/bCNC/CNCRibbon.py", line 143, in addRibbonGroup
    self.ribbons.append((Page.groups[name], args))
KeyError: 'Generator'

smooker, 2017-01-20:
make clean

closed 2017-01-20 by smooker

vlachoudis, 2017-01-20:
There is a patch on the new master, that it doesn't stop but reports a warning to clean up the .bCNC file

smooker, 2017-01-24:
Thanks!

----- Цитат от Vasilis Vlachoudis (notifications@github.com), на
20.01.2017 в 15:27 -----  

There is a patch on the new master, that it doesn't stop but reports a
warning to clean up the .bCNC file 

—
[2].   
	*      

Links:
------
[1] https://github.com/vlachoudis/bCNC/issues/498#issuecomment-274071722
[2]

-------------------------------------
Mail.BG: Безплатен e-mail адрес. Най-добрите характеристики на българския пазар - 30 GB пощенска кутия, 1 GB прикрепен файл, безплатен POP3, мобилна версия и други. http://mail.bg