vlachoudis/bCNC #502: Configuration GRBL 1.1e

robcop2000, 2017-01-22:
After the upgrade to GRBL 1.1e I had some problem.
First the old electronic make strange behaviors, so I have changed everyting, Arduino, Stepper motors, CNC shield etc...All new.
Now the configuration has strange behaviors.
1 - The $3 – Direction port invert, mask doen't change any value I put inside.
2- I have the same motor, driver on the CNC shield both are set to 1/8 for the X and Y axis. Why I have to insert in $100, $101 [X,Y] steps/mm two different value to have the correct movement?
3- Now the CNC controller saies that the GRBL is no connected, but if I made jog movements it works?

In GRBL forum they say is a controller problem

chamnit, 2017-01-22:
@robcop2000 : Lets figure out if it's something firmware related first. bCNC supports Grbl v1.1 fairly well.

vlachoudis, 2017-01-22:
@robcop2000 Did you try to make a complete reset on the eeprom of grbl?

robcop2000, 2017-01-22:
I have changes the Arduino at all. All Electronic has changed with a new one (arduino Uno, stepper driver, GAUPS board).

vlachoudis, 2017-01-23:
@robcop2000 You have a few components to check
bCNC -> Grbl -> Arduino -> CNC Shield -> Motor Drivers -> Stepper motors -> Machine
bCNC you can easily check. Send a command and see if grbl returns ok in the terminal. Or open a terminal on your system and send manually the commands.
This will exclude that is the GUI as the problem which seems so with the symptoms you write.
My guess is that since you changed your hardware completely I believe the problem is somewhere there. Swap the drivers on X/Y and see the behaviour, try to move reduce the speed, acceleration.

robcop2000, 2017-01-23:
Now the situation is this.
I Have flashed the Arduino board on the PC and configurated.
I have changed the USB cable ( Was the only thing old)
I have installed on the machine an seemes to work.

The only problem is Homing, that first the Z axis runs in Home position, after stops when Spindle reach the X or Y limit, and not both like before.

robcop2000, 2017-01-23:
I don't Think was a bCNC problem I have used USG and the problem was the same.

robcop2000, 2017-01-23:
My configuration is this
GRBL 1.1e Config

$0=10
$1=25
$2=0
$3=4
$4=0
$5=0
$6=0
$10=1
$11=0.010
$12=0.002
$13=0
$20=0
$21=1
$22=1
$23=3
$24=25.000
$25=500.000
$26=250
$27=1.000
$30=1000
$31=0
$32=0
$100=80.000
$101=40.000
$102=400.000
$110=2000.000
$111=2000.000
$112=200.000
$120=500.000
$121=500.000
$122=50.000
$130=830.000
$131=1830.000
$132=170.000

chamnit, 2017-01-23:
@robcop2000 : Given that it does the same thing in UGS as in bCNC and you were able to fix it with changing out the USB cable, this isn't a bCNC problem. I would close this and continue with your currently open Grbl issue thread.

closed 2017-01-24 by vlachoudis