vlachoudis/bCNC #504: Can't copy grbl settings

poesel, 2017-01-26:
Under Linux I can't copy the grbl output in the terminal window. Which would be useful for backup. 
Or as an improvement allow an export to file right there.

jeffmd, 2017-01-27:
I just left click in the terminal area to highlight the lines I want to copy, then do Ctrl-c to copy the selected text.  Then I open up my favourite text editor and do Ctrl-v to paste the selected text.  Works on my Raspberry Pi.    Using Ctrl-insert for copy and Shift-insert to paste also works.  Works the same on Fedora too.

poesel, 2017-01-27:
Hmm, I tried again and now it works. But I also updated from 0.95 to 0.97.
Sorry - should have tested better. This issue can be closed.

closed 2017-01-27 by poesel