vlachoudis/bCNC #505: Adjusting Scale for Higher DPI Screens

magveda, 2017-01-26:
Is it possible to change the DPI settings for high resolution screen?

vlachoudis, 2017-01-27:
From bCNC you can change the fonts in the Tab: Tools -> Fonts, but not the DPI.
Now in the font size if you set a negative number e.g. -12, tk creates the font to the exact pixel size,
with positive 12, it will use the DPI of your screen.

Effer, 2017-01-28:
Doubling icon size on load may also help. 
You can try by modify the method **loadIcons()**  in the file Utilis.py 

 ```python
def loadIcons():
	global icons
	icons = {}
	for img in glob.glob("%s%sicons%s*.gif"%(prgpath,os.sep,os.sep)):
		name,ext = os.path.splitext(os.path.basename(img))
		try:
			icons[name] = PhotoImage(file=img)
			icons[name] = icons[name].zoom(2, 2) #<--  Double size Icon
		except TclError:
			pass

magveda, 2017-01-28:
Thanks! Will try these suggestions!

vlachoudis, 2017-01-28:
@Effer it should be added as a user preference the zooming of icons

Effer, 2017-01-28:
Ok will do.

vlachoudis, 2017-01-29:
Thanks to @Effer it is now an option in the preferences. #506

closed 2017-01-29 by vlachoudis

magveda, 2017-01-31:
You are great!

garyemiller, 2023-12-22:
Help!  Where is that config option?  I can't find it.

Here is a screen shot, so you see how hard it is to read:

[image]