vlachoudis/bCNC #508: Problem Importing DXF files

AlvaroRey, 2017-01-30:
Hi, first its first, Thanks for the program ;)

My problem, I tried to upload a few files, I use corel draw to create the drawings, and export diferent dxf formats, but when I try to open it in Bcnc, the circles seam like squeres with round corners

Thanks in advance

vlachoudis, 2017-01-30:
Hi @AlvaroRey could you post your dxf file to check. Just a tip, bCNC doesn't support blocks for the moment, so try to explode them before importing.

AlvaroRey, 2017-01-30:
https://www.dropbox.com/s/ie0l94nz20ir4qc/Logo%20MakerGal.dxf?dl=0 that's the file

vlachoudis, 2017-01-31:
I don't understand, on my version it loads the dxf file and displays exactly the same as in librecad
[image]
[image]

AlvaroRey, 2017-01-31:
When I open the files in my computer its completly round. Its made with
circles. I will upload a picture from the software

vlachoudis, 2017-02-01:
Indeed I've test it with FreeCAD and it shows as circles. I will check the implementation of the splines, maybe I miss some parameter

vlachoudis, 2017-02-02:
I've found the problem, I am experimenting on how to fix it. I was ignoring the knotsvector from the dxf.

AlvaroRey, 2017-02-02:
This is how I sew it in my software, i forgot to send it yesterday

vlachoudis, 2017-02-06:
I've pushed a patch on the current master, that tries to correct the problem, and it works for your case.
However I am not fully sure that it works for all cases. The problem is that I cannot find any reasonable documentation on what are all the options in the dxf spline, how to handle the periodic vs closed, etc...
As well I added the ability to handle dxf blocks (tentatively).

closed 2017-02-06 by vlachoudis

AlvaroRey, 2017-02-06:
Ok perfect thanks, I will tested