vlachoudis/bCNC #509: Feedrate overide with Smoothie

brotherbrown831, 2017-01-31:
I am using the most recent version of bcnc and smoothieware cnc firmware. Basic operation and job execution are working great but the real time feedrate adjustment does not seem to have any effect either prior to the start of a job or during its execution. I have enable "grbl_mode True" in my firmware and connected to my board via both the "smoothie" option in the drop down and the grbl-v0 option with no change in behavior. 

Thank you for the great work!

vlachoudis, 2017-01-31:
I will give a look later this week to see if something is broken

brotherbrown831, 2017-01-31:
If there are any logs or dumps I can provide that would assist let me know.

brotherbrown831, 2017-02-04:
Any luck with this?

vlachoudis, 2017-02-05:
I've just pushed an update to fix the issue for smoothie. I forgot to add a condition for smoothie when I've made the changes for Grbl v1+

closed 2017-02-05 by vlachoudis